rst-tu-dortmund/teb_local_planner #182: Build error in the ROS Kinetic version

kay01-kwon, 2019-11-13:
When building the teb_local_planner package by typing catkin_make in the ~/catkin_ws directory, the compiler cannot find out a lot of header files.

The error message is the following : 

/home/kay/catkin_ws/src/Dual_arm_project/teb_local_planner/src/optimal_planner.cpp: In member function ‘boost::shared_ptr<g2o::SparseOptimizer> teb_local_planner::TebOptimalPlanner::initOptimizer()’:
/home/kay/catkin_ws/src/Dual_arm_project/teb_local_planner/src/optimal_planner.cpp:164:91: error: no matching function for call to ‘g2o::BlockSolver<g2o::BlockSolverTraits<-1, -1> >::BlockSolver(std::remove_reference<std::unique_ptr<g2o::LinearSolverCSparse<Eigen::Matrix<double, -1, -1> > >&>::type)’
   std::unique_ptr<TEBBlockSolver> block_solver(new TEBBlockSolver(std::move(linear_solver)));
                                                                                           ^
In ROS Melodic version, the build process works well. I guess that because the compiler cannot locate header files in ROS Kinetic version, the member function is not declared. How can you fix this problem in the ROS kinetic version?

croesmann, 2019-11-13:
It seams that you are still trying to build the melodic version, since the error message dispays the following code 

```
std::unique_ptr block_solver(new TEBBlockSolver(std::move(linear_solver)));
```
which is melodic code and not in kinetic.
Have you switched branches properly after cloning? E.g. `git checkout kinetic-devel`

kay01-kwon, 2019-11-13:
Yes, I checked out the branch belongs to kinetic-devel.

kay01-kwon, 2019-11-13:
Oh, when copying the address and git clone the address, melodic version is
downloaded. By downloading zip file, the problem is solved. Thanks.

2019년 11월 13일 (수) 오후 10:08, ­권건우(대학원/일반대학원 기계공학과) <kay1216@yonsei.ac.kr>님이
작성:

croesmann, 2019-11-13:
Git always clones the default branch if you do not specify any specific branch, e.g. like

`git clone -b kinetic-devel https://github.com/rst-tu-dortmund/teb_local_planner`

Otherwise you can switch branches afterwards as mentioned in my previous comment.

Closing this now.

closed 2019-11-13 by croesmann