rst-tu-dortmund/teb_local_planner #192: Create ROS2 Master Branch

SteveMacenski, 2020-02-20:
Hi,

Right now, with only the eloquent / dashing branches, TEB is falling behind the Navigation2 interfaces quite a bit. As Nav2 is charging ahead (unlike navigation in ROS1), the migration between distributions to do all at once will be challenging and error prone. Moreover, a lot of people are running navigation2 master branch because of how many features and bug fixes they'd be missing on the older distribution branches, while they're still ABI compatible (Eloquent, mostly). 

Especially since we're chatting about including TEB as a default controller. Can you create a ROS2 master branch so we can make incremental updates as required? What are your plans to keep ROS1 and ROS2 in sync (or at least make sure ROS2 gets the ROS1 updates)?

croesmann, 2020-02-20:
Hi,

good point, I am not used to such frequent changes in the navigation API ;-).
I create `ros2-master` branch.

Currently, my plan is to merge ROS1 features into ros2-master and resolve conflicts manually.
But I am open to other best practices!

SteveMacenski, 2020-02-20:
That's more or less best practice, but we usually use the git cherry pick, but it really depends on how picky you are about your git history (I'm not picky at all)

SteveMacenski, 2020-02-20:
Awesome - thanks for the branch, the only other thing I'd want for the ros2-master is changing the `penalty_epsilon` from 0.1 -> 0.05 per https://github.com/rst-tu-dortmund/teb_local_planner/issues/28, assuming that makes sense in your mind as a default. Having a good "out of box experience" and all

croesmann, 2020-02-20:
I'll sync with ros1 tomorrow morning. And sure, we can update that parameter!

SteveMacenski, 2020-02-20:
There's not enough emojis to describe how helpful that is. 

Are you a ROS2 user yet? Its the face of things to come & stable enough to really dive into as of Eloquent. I wouldn't develop any new code in ROS1 anymore unless you can't help it.

croesmann, 2020-02-21:
done ;-)

Actually, not yet. We use turtlebot2 robots for teaching etc. and these are not even fully supported in melodic yet. But yeah, the progress of ros2 and especially navigation2 looks very good! ;-)

closed 2020-02-28 by SteveMacenski