rst-tu-dortmund/teb_local_planner #195: when I use  TebOptimalPlanner::plan(const PoseSE2& start, const PoseSE2& goal, const geometry_msgs::Twist* start_vel, bool free_goal_vel) ,I have some question

xujiangfan, 2020-03-04:
Hi
  I downloaded the latest code, compiled and debugged it, and found that the stopping accuracy was much worse than my setting;
I modified the plan function, as the title;
Although the final accuracy meets the requirements, the shock is obvious;
Is there any way to make the movement trajectory approach the end slowly without oscillation;

 xy_goal_tolerance: 0.01
 yaw_goal_tolerance: 0.01

by the way,
in teb_local_planner_ros.cpp line 323:
  // Now perform the actual planning
  bool success = planner_->plan(robot_pose_, robot_goal_, &robot_vel_, cfg_.goal_tolerance.free_goal_vel); // straight line init
  //bool success = planner_->plan(transformed_plan, &robot_vel_, cfg_.goal_tolerance.free_goal_vel);

You need to add & before robot_vel_, otherwise code won't compile.

croesmann, 2020-03-04:
Hi, 
please avoid such verbose and confusing issue titles.
Are you sure that you cloned the correct branch?
Line 323 differs from your line and the call to planner_->plan() is correct and working in `melodic-devel` branch.

Meroke, 2025-07-11:
I have the same issue. However, I found that 0.01m accuracy is unachievable for most radars—indoor single-line radars typically offer only 0.03~0.05m precision. Relying solely on radar for navigation makes high-accuracy positioning difficult.
To improve target-reaching precision, I reduce linear/angular speeds and split motions (e.g., rotate first, then move at corners) to stabilize radar-based localization. 
Yet, probabilistic fluctuations persist, causing significant deviations from targets.
I’m also exploring better solutions.