rst-tu-dortmund/teb_local_planner #196: [ROS2 Master] SegFault on printErrorWhenNotInitialized()

maxlein, 2020-03-05:
Using the current ros2-master branch I get a segfault.  
Looks like the node when using the logger for the error message is not set:  

```
Thread 1 "controller_serv" received signal SIGSEGV, Segmentation fault.
0x0000555555b50a10 in ?? ()
(gdb) bt
#0  0x0000555555b50a10 in  ()
#1  0x00007ffff49b06de in  () at /lib/x86_64-linux-gnu/libgcc_s.so.1
#2  0x00007ffff49b0ef5 in _Unwind_Resume () at /lib/x86_64-linux-gnu/libgcc_s.so.1
#3  0x00007fffba6060f3 in teb_local_planner::TebVisualization::printErrorWhenNotInitialized() const [clone .cold.991] () at /home/kin/alpen/ros2_ws/install/teb_local_planner/lib/libteb_local_planner.so
#4  0x00007ffff78d2c03 in rclcpp::callback_group::CallbackGroup::add_subscription(std::shared_ptr<rclcpp::SubscriptionBase>) () at /opt/ros/eloquent/lib/librclcpp.so
#5  0x00007ffff4fe9258 in operator new(unsigned long) () at /usr/lib/x86_64-linux-gnu/libstdc++.so.6
#6  0xffe8c5ebf5b53900 in  ()
#7  0x00007fffffff7880 in  ()
#8  0x00005555558f3940 in  ()
#9  0x00007fffffff6fc0 in  ()
#10 0x00007fffffff6f80 in  ()
#11 0x00007fffffff7880 in  ()
#12 0x00007fffffff8117 in  ()
#13 0x00007fffffff8270 in  ()
#14 0x00007ffff7bc9658 in rclcpp_lifecycle::LifecycleNode::LifecycleNodeInterfaceImpl::execute_callback(unsigned int, rclcpp_lifecycle::State const&) () at /opt/ros/eloquent/lib/librclcpp_lifecycle.so
#15 0x00007ffff7bc9aa1 in rclcpp_lifecycle::LifecycleNode::LifecycleNodeInterfaceImpl::change_state(unsigned char, rclcpp_lifecycle::node_interfaces::LifecycleNodeInterface::CallbackReturn&) () at /opt/ros/eloquent/lib/librclcpp_lifecycle.so
#16 0x00007ffff7bca0b5 in rclcpp_lifecycle::LifecycleNode::LifecycleNodeInterfaceImpl::on_change_state(std::shared_ptr<rmw_request_id_t>, std::shared_ptr<lifecycle_msgs::srv::ChangeState_Request_<std::allocator<void> > >, std::shared_ptr<lifecycle_msgs::srv::ChangeState_Response_<std::allocator<void> > >) () at /opt/ros/eloquent/lib/librclcpp_lifecycle.so
#17 0x00007ffff7bc388d in std::_Function_handler<void (std::shared_ptr<rmw_request_id_t>, std::shared_ptr<lifecycle_msgs::srv::ChangeState_Request_<std::allocator<void> > >, std::shared_ptr<lifecycle_msgs::srv::ChangeState_Response_<std::allocator<void> > >), std::_Bind<void (rclcpp_lifecycle::LifecycleNode::LifecycleNodeInterfaceImpl::*(rclcpp_lifecycle::LifecycleNode::LifecycleNodeInterfaceImpl*, std::_Placeholder<1>, std::_Placeholder<2>, std::_Placeholder<3>))(std::shared_ptr<rmw_request_id_t>, std::shared_ptr<lifecycle_msgs::srv::ChangeState_Request_<std::allocator<void> > >, std::shared_ptr<lifecycle_msgs::srv::ChangeState_Response_<std::allocator<void> > >)> >::_M_invoke(std::_Any_data const&, std::shared_ptr<rmw_request_id_t>&&, std::shared_ptr<lifecycle_msgs::srv::ChangeState_Request_<std::allocator<void> > >&&, std::shared_ptr<lifecycle_msgs::srv::ChangeState_Response_<std::allocator<void> > >&&) () at /opt/ros/eloquent/lib/librclcpp_lifecycle.so
#18 0x00007ffff7bc477c in rclcpp::Service<lifecycle_msgs::srv::ChangeState>::handle_request(std::shared_ptr<rmw_request_id_t>, std::shared_ptr<void>) () at /opt/ros/eloquent/lib/librclcpp_lifecycle.so
#19 0x00007ffff78e06a0 in rclcpp::executor::Executor::execute_service(std::shared_ptr<rclcpp::ServiceBase>) () at /opt/ros/eloquent/lib/librclcpp.so
#20 0x00007ffff78e10a5 in rclcpp::executor::Executor::execute_any_executable(rclcpp::executor::AnyExecutable&) () at /opt/ros/eloquent/lib/librclcpp.so
#21 0x00007ffff78e771f in rclcpp::executors::SingleThreadedExecutor::spin() () at /opt/ros/eloquent/lib/librclcpp.so
#22 0x00007ffff78e4412 in rclcpp::spin(std::shared_ptr<rclcpp::node_interfaces::NodeBaseInterface>) () at /opt/ros/eloquent/lib/librclcpp.so

```

dkuenster, 2020-05-15:
Getting the same error with ros2_master and the current ros2 and nav2 master.

dkuenster, 2020-05-19:
It doesn't crash if you build it in Debug Mode.

3wnbr1, 2020-06-21:
I used to have the issue when building with RelWithDebInfo
```
Stopped reason: SIGSEGV

#0  0x00007ff2fd8aed27 in  () at /lib/x86_64-linux-gnu/libgcc_s.so.1
#1  0x00007ff2fd8af615 in _Unwind_Resume () at /lib/x86_64-linux-gnu/libgcc_s.so.1
#2  0x00007ff2f82c2a7b in std::__shared_count<(__gnu_cxx::_Lock_policy)2>::~__shared_count() (this=<optimized out>, __in_chrg=<optimized out>) at /usr/include/c++/9/bits/shared_ptr_base.h:1169
#3  std::__shared_ptr<std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const, (__gnu_cxx::_Lock_policy)2>::~__shared_ptr() (this=<optimized out>, __in_chrg=<optimized out>)
    at /usr/include/c++/9/bits/shared_ptr_base.h:1169
#4  std::shared_ptr<std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const>::~shared_ptr() (this=<optimized out>, __in_chrg=<optimized out>) at /usr/include/c++/9/bits/shared_ptr.h:103
#5  rclcpp::Logger::~Logger() (this=<optimized out>, __in_chrg=<optimized out>) at /opt/ros/foxy/include/rclcpp/logger.hpp:77
#6  teb_local_planner::TebVisualization::printErrorWhenNotInitialized() const (this=0x559b6beb91c0) at /home/vagrant/ros_ws/teb_local_planner/teb_local_planner/src/visualization.cpp:500
#7  0x00021f0e095b0f01 in  ()
#8  0x0000000000000000 in  ()
```

Great work @dkuenster, after applying the patch, it works just fine

croesmann, 2020-06-22:
thanks, this should now be fixed as #212 has been merged.

closed 2020-06-22 by croesmann