rst-tu-dortmund/teb_local_planner #200: velocity command disappears for a while

zengxiaolei, 2020-03-26:
Hi, Christoph,
I use TEB-local-planner in ROS melodic (0.8.4), Ubuntu 18.04. 
As shown in Fig. 1, velocity command from local planner disappears for a while at time 2260.5s and  at 2264.3s, and velocity frequency is relative low after these time-points for some time, which results in strange local plan: the local path can form one or more circles.

[image]

Fig. 1
move_base_vel : original velocity topic of move_base
cmd_vel : smoothed move_base_vel
odom: odom velocity feedback

This phenomenon can happens, especially at start point or end point. I don't know if this is a problem caused by inappropriate parameters or the algorithm itself. I look forward to your help.

adamkrawczyk, 2020-04-21:
I think, I have a similar problem when robot moves sometimes planner stops sending cmd_vel data and robot uses last published velocity. When this happens I can see warnings that `control loop missed desired frequency`. I was wondering if this can be caused by not enough computation power but I'm using 16cpu and usage is about 50%.

srsidd, 2020-08-05:
I'm seeing a similar problem, except I never see the planner recover.. Were you able to figure this out?

adamkrawczyk, 2020-08-06:
Unfortunately I haven't and changed to other local planner.

zengxiaolei, 2020-08-07:
Hi, I've solved the problem. In my case,  the reason is that the global plan path is not smooth. After fix global path parameter, the problem is solved.