rst-tu-dortmund/teb_local_planner #203: eloquent - planner jumps to goal pose

tgreier, 2020-04-13:
Hello,
I have an issue that I hope you can explain. I am trying to use TEB planner in scenario where I want the local planner to closely follow the global plan. The global plan specifies orientations for each via point. I have read tutorials on how to do this using via-points and increasing the weight for via points, but I still experience a situation where the planner will ignore the global plan and plans a direct path to the goal pose.  The attached file contains a movie showing the behavior. I have tried decreasing my look-ahead as well, but the issue still occurs. I also have shortest path and optimal time weights set to 0.0. I tried setting them to a small value instead of 0, but no effect.

Config file is below.
Thank you for your help.

controller_server:
  ros__parameters:
    use_sim_time: True

    # Controller Server
    controller_frequency: 20.0
    controller_plugin_ids: ["FollowPath"]
    controller_plugin_types: ["teb_local_planner::TebLocalPlannerROS"]
    min_x_velocity_threshold: 0.0001
    min_y_velocity_threshold: 0.0001
    min_theta_velocity_threshold: 0.0001

    # Progress Checker
    required_movement_radius: 0.25
    movement_time_allowance: 20.0

    FollowPath:
      # TEB: Configuration parameter descriptions: https://github.com/rst-tu-dortmund/teb_local_planner/blob/eloquent-devel/teb_local_planner/include/teb_local_planner/teb_config.h
      odom_topic: odom
      map_frame: /map
          
      # Trajectory
        
      teb_autosize: 1.0
      dt_ref: 0.3
      dt_hysteresis: 0.1
      min_samples: 3
      max_samples: 500
      global_plan_overwrite_orientation: True
      allow_init_with_backwards_motion: False
      global_plan_viapoint_sep: 0.5
      via_points_ordered: True
      max_global_plan_lookahead_dist: 2.0
      global_plan_prune_distance: 1.0
      exact_arc_length: False
      force_reinit_new_goal_dist: 1.0
      feasibility_check_no_poses: 5
      publish_feedback: False
      min_resolution_collision_check_angular: 3.14159265358979323846264338327950288

      # Robot
              
      max_vel_x: 0.4
      max_vel_x_backwards: 0.0
      max_vel_y: 0.0
      max_vel_theta: 0.3
      acc_lim_x: 0.5
      acc_lim_y: 0.5
      acc_lim_theta: 0.5
      min_turning_radius: 0.5
      wheelbase: 1.0
      cmd_angle_instead_rotvel: False
      is_footprint_dynamic: False

      footprint_model: # types: "point", "circular", "two_circles", "line", "polygon" (avoid polygon as higher CPU)
        type: "line"
        radius: 0.2 # for type "circular"
        line_start: [-0.5, 0.0] # for type "line"
        line_end: [0.5, 0.0]
        front_offset: 0.2 # type "two_circles"
        front_radius: 0.2
        rear_offset: 0.2
        rear_radius: 0.2
        # vertices: [ [0.25, -0.05], [0.18, -0.05], [0.18, -0.18], [-0.19, -0.18], [-0.25, 0], [-0.19, 0.18], [0.18, 0.18], [0.18, 0.05], [0.25, 0.05] ] # for type "polygon"
        # this type is not yet parsing properly in ROS2 rclcpp

      # GoalTolerance
          
      xy_goal_tolerance: 0.2
      yaw_goal_tolerance: 0.1
      free_goal_vel: False
      complete_global_plan: True
          
      # Obstacles
          
      min_obstacle_dist: 0.4 # This value must also include our robot's expansion, since footprint_model is set to "line".
      inflation_dist: 0.6
      dynamic_obstacle_inflation_dist: 0.6
      include_dynamic_obstacles: True
      include_costmap_obstacles: True
      costmap_obstacles_behind_robot_dist: 1.0
      obstacle_poses_affected: 15
      legacy_obstacle_association: False
      obstacle_association_force_inclusion_factor: 1.5
      obstacle_association_cutoff_factor: 5.0
      costmap_converter_plugin: ""
      costmap_converter_spin_thread: True
      costmap_converter_rate: 5.0

      # Optimization
          
      no_inner_iterations: 5
      no_outer_iterations: 4
      optimization_activate: True
      optimization_verbose: False
      penalty_epsilon: 0.1
      obstacle_cost_exponent: 4.0
      weight_max_vel_x: 2.0
      weight_max_vel_y: 2.0
      weight_max_vel_theta: 1.0
      weight_acc_lim_x: 1.0
      weight_acc_lim_y: 1.0
      weight_acc_lim_theta: 1.0
      weight_kinematics_nh: 1000.0
      weight_kinematics_forward_drive: 1000.0
      weight_kinematics_turning_radius: 1.0
      weight_optimaltime: 0.0
      weight_shortest_path: 0.0
      weight_obstacle: 100.0
      weight_inflation: 0.2
      weight_dynamic_obstacle: 10.0
      weight_dynamic_obstacle_inflation: 0.2
      weight_viapoint: 100.0
      weight_prefer_rotdir: 1.0
      weight_adapt_factor: 2.0
      
      # Homotopy Class Planner

      enable_homotopy_class_planning: True
      enable_multithreading: True
      simple_exploration: False
      max_number_classes: 4
      selection_obst_cost_scale: 1.0
      selection_prefer_initial_plan: 0.95
      selection_viapoint_cost_scale: 1.0
      selection_cost_hysteresis: 1.0
      selection_alternative_time_cost: False
      switching_blocking_period: 0.0
      roadmap_graph_samples: 15
      roadmap_graph_area_width: 5.0
      roadmap_graph_area_length_scale: 1.0
      h_signature_prescaler: 0.5
      h_signature_threshold: 0.1
      obstacle_keypoint_offset: 0.1
      obstacle_heading_threshold: 0.45
      viapoints_all_candidates: True
      visualize_hc_graph: False
      visualize_with_time_as_z_axis_scale: 0.0
      delete_detours_backwards: True
      detours_orientation_tolerance: 1.570796327
      length_start_orientation_vector: 0.4
      max_ratio_detours_duration_best_duration: 3.0

      # Recovery
      shrink_horizon_backup: True
      shrink_horizon_min_duration: 10.0
      oscillation_recovery: True
      oscillation_v_eps: 0.1
      oscillation_omega_eps: 0.1
      oscillation_recovery_min_duration: 10.0
      oscillation_filter_duration: 10.0

[jumptogoalpose.zip](https://github.com/rst-tu-dortmund/teb_local_planner/files/4472038/jumptogoalpose.zip)

tgreier, 2020-04-13:
I found this pull request, which seems to resolve my problem: https://github.com/rst-tu-dortmund/teb_local_planner/pull/198

jjKeyLite, 2020-09-27:
I experienced the same problem. After checking the algorithm, I found that in function: transformGlobalPlan(), when the transformed_plan is empty, the algorithm will push the final global goal back to it. In this case, the goal of teb planner will be the global goal, which your mentioned "jumping to goal pose" happened.
 So check the process in function:transformGlobalPlan() whether something wrong happened that resulting in transformed_plan to be empty.

SteveMacenski, 2021-03-23:
#291 resolves

closed 2021-03-23 by SteveMacenski