rst-tu-dortmund/teb_local_planner #206: may you explain why you use fast sigmoid in acceleration edge?

yinzixuan126, 2020-05-17:
// consider directions
//     vel1 *= g2o::sign(diff1[0]*cos(pose1->theta()) + diff1[1]*sin(pose1->theta())); 
//     vel2 *= g2o::sign(diff2[0]*cos(pose2->theta()) + diff2[1]*sin(pose2->theta())); 
    vel1 *= fast_sigmoid( 100*(diff1.x()*cos(pose1->theta()) + diff1.y()*sin(pose1->theta())) ); 
    vel2 *= fast_sigmoid( 100*(diff2.x()*cos(pose2->theta()) + diff2.y()*sin(pose2->theta())) );

XiruiLiu038, 2020-07-15:
To my personal of view ,the purpose is to ensure that the function of velocity is proximately continuous and differentiable.

croesmann, 2020-07-15:
The answer of @XiruiLiu038 is correct. Furthermore, I use the `sign/sigmoid` function to determine the current direction of travel.

closed 2020-07-15 by croesmann