rst-tu-dortmund/teb_local_planner #211: Planner crash in during trajectory on ROS Melodic from src

tomlogan501, 2020-06-03:
Hi,
I tested the planner on the online tutorial of robotignite with the husky : https://www.robotigniteacademy.com/en/course/teb-local-planner/details/

Our lab robot is also a husky with various sensors, I transferred all the parameters to my Gazebo simulation from robotignite tutorial.

I have various problems such as missed loop updates on the simulation.
I tested the optim node and realize that the parameter /test_optim_node/enable_homotopy_class_planning shutdown the optim node when I started playing around on RVIZ.

```
tsimon@GL503GE:~/teb_catkin_ws$ rosparam set /test_optim_node/enable_homotopy_class_planning True
/opt/ros/melodic/lib/python2.7/dist-packages/rosparam/__init__.py:370: YAMLLoadWarning: calling yaml.load() without Loader=... is deprecated, as the default Loader is unsafe. Please read https://msg.pyyaml.org/load for full details.
  set_param_raw(param, yaml.load(value), verbose=verbose)
tsimon@GL503GE:~/teb_catkin_ws$ roslaunch teb_local_planner test_optim_node.launch
... logging to /home/tsimon/.ros/log/ee5d56fe-a572-11ea-a3eb-91d3f2cf57d9/roslaunch-GL503GE-29609.log
Checking log directory for disk usage. This may take awhile.
Press Ctrl-C to interrupt
Done checking log file disk usage. Usage is <1GB.

started roslaunch server http://GL503GE:40715/

SUMMARY
========

PARAMETERS
 * /rosdistro: melodic
 * /rosversion: 1.14.3

NODES
  /
    rviz (rviz/rviz)
    test_optim_node (teb_local_planner/test_optim_node)

ROS_MASTER_URI=http://localhost:11311

process[test_optim_node-1]: started with pid [29635]
process[rviz-2]: started with pid [29642]
[ INFO] [1591174894.346716833]: No robot footprint model specified for trajectory optimization. Using point-shaped model.
[test_optim_node-1] process has died [pid 29635, exit code -11, cmd /home/tsimon/teb_catkin_ws/devel/lib/teb_local_planner/test_optim_node __name:=test_optim_node __log:=/home/tsimon/.ros/log/ee5d56fe-a572-11ea-a3eb-91d3f2cf57d9/test_optim_node-1.log].
log file: /home/tsimon/.ros/log/ee5d56fe-a572-11ea-a3eb-91d3f2cf57d9/test_optim_node-1*.log

```

Here is the parameter list for the husky

```
controller_frequency: 3.0
recovery_behaviour_enabled: true

NavfnROS:
  allow_unknown: true # Specifies whether or not to allow navfn to create plans that traverse unknown space.
  default_tolerance: 0.1 # A tolerance on the goal point for the planner.

TebLocalPlannerROS:

 odom_topic: odometry/filtered
    
 # Trajectory
  
 teb_autosize: True
 dt_ref: 0.3
 dt_hysteresis: 0.1
 global_plan_overwrite_orientation: True
 allow_init_with_backwards_motion: False
 max_global_plan_lookahead_dist: 3.0
 feasibility_check_no_poses: 5
    
 # Robot
         
 max_vel_x: 2.0
 max_vel_x_backwards: 2.2
 max_vel_y: 0.0
 max_vel_theta: 2.0
 acc_lim_x: 2.5
 acc_lim_theta: 2.5
 min_turning_radius: 0.0 # diff-drive robot (can turn on place!)

 footprint_model:
   type: "point"

 # GoalTolerance
    
 xy_goal_tolerance: 0.2
 yaw_goal_tolerance: 0.1
 free_goal_vel: False
    
 # Obstacles
    
 min_obstacle_dist: 0.75 # This value must also include our robot radius, since footprint_model is set to "point".
 include_costmap_obstacles: True
 costmap_obstacles_behind_robot_dist: 1.5
 obstacle_poses_affected: 30
 costmap_converter_plugin: ""
 costmap_converter_spin_thread: True
 costmap_converter_rate: 5

 # Optimization
    
 no_inner_iterations: 5
 no_outer_iterations: 4
 optimization_activate: True
 optimization_verbose: False
 penalty_epsilon: 0.1
 weight_max_vel_x: 2
 weight_max_vel_theta: 1
 weight_acc_lim_x: 1
 weight_acc_lim_theta: 1
 weight_kinematics_nh: 1000
 weight_kinematics_forward_drive: 1
 weight_kinematics_turning_radius: 1
 weight_optimaltime: 1
 weight_obstacle: 50
 weight_dynamic_obstacle: 10 # not in use yet
 weight_adapt_factor: 2

 # Homotopy Class Planner

 enable_homotopy_class_planning: True
 enable_multithreading: True
 simple_exploration: False
 max_number_classes: 4
 selection_cost_hysteresis: 1.0
 selection_obst_cost_scale: 1.0
 selection_alternative_time_cost: False
 
 roadmap_graph_no_samples: 15
 roadmap_graph_area_width: 5
 h_signature_prescaler: 0.5
 h_signature_threshold: 0.1
 obstacle_keypoint_offset: 0.1
 obstacle_heading_threshold: 0.45
 visualize_hc_graph: False
```

Do you have a docker file to check everything on melodic ? 
Should I check other parameters due to kinetic version on robotignite ?

tomlogan501, 2020-06-03:
I made a fresh update of the source code use with 
`rosinstall_generator teb_local_planner --rosdistro melodic --deps --tar  --deps-depth 2 > teb_planner.rosinstall`

No changes

tomlogan501, 2020-06-03:
Seems to be g2o or boost 

```
(gdb) backtrace
#0  __GI___libc_free (mem=0x65) at malloc.c:3102
#1  0x00007ffff758c538 in g2o::OptimizableGraph::~OptimizableGraph() () from /usr/local/lib/libg2o_core.so
#2  0x00007ffff7ecdad8 in boost::detail::sp_counted_impl_pd<g2o::SparseOptimizer*, boost::detail::sp_ms_deleter<g2o::SparseOptimizer> >::dispose() ()
   from /home/tsimon/teb_catkin_ws/devel/lib/libteb_local_planner.so
#3  0x00007ffff7ecd702 in teb_local_planner::TebOptimalPlanner::~TebOptimalPlanner() () from /home/tsimon/teb_catkin_ws/devel/lib/libteb_local_planner.so
#4  0x00007ffff7ecd79d in teb_local_planner::TebOptimalPlanner::~TebOptimalPlanner() () from /home/tsimon/teb_catkin_ws/devel/lib/libteb_local_planner.so
#5  0x00005555555a302a in boost::detail::sp_counted_base::release() ()
#6  0x00007ffff7f11640 in teb_local_planner::HomotopyClassPlanner::renewAndAnalyzeOldTebs(bool) () from /home/tsimon/teb_catkin_ws/devel/lib/libteb_local_planner.so
#7  0x00007ffff7f11e30 in teb_local_planner::HomotopyClassPlanner::exploreEquivalenceClassesAndInitTebs(teb_local_planner::PoseSE2 const&, teb_local_planner::PoseSE2 const&, double, geometry_msgs::Twist_<std::allocator<void> > const*) () from /home/tsimon/teb_catkin_ws/devel/lib/libteb_local_planner.so
#8  0x00007ffff7f11f7c in teb_local_planner::HomotopyClassPlanner::plan(teb_local_planner::PoseSE2 const&, teb_local_planner::PoseSE2 const&, geometry_msgs::Twist_<std::allocator<void> > const*, bool) ()
   from /home/tsimon/teb_catkin_ws/devel/lib/libteb_local_planner.so
#9  0x000055555559749d in CB_mainCycle(ros::TimerEvent const&) ()
#10 0x00007ffff7bbd19f in ros::TimerManager<ros::Time, ros::Duration, ros::TimerEvent>::TimerQueueCallback::call() () from /opt/ros/melodic/lib/libroscpp.so
#11 0x00007ffff7bdf018 in ros::CallbackQueue::callOneCB(ros::CallbackQueue::TLS*) () from /opt/ros/melodic/lib/libroscpp.so
#12 0x00007ffff7bdfcd3 in ros::CallbackQueue::callAvailable(ros::WallDuration) () from /opt/ros/melodic/lib/libroscpp.so
#13 0x00007ffff7c32809 in ros::SingleThreadedSpinner::spin(ros::CallbackQueue*) () from /opt/ros/melodic/lib/libroscpp.so
#14 0x00007ffff7c1b4bf in ros::spin() () from /opt/ros/melodic/lib/libroscpp.so
#15 0x00005555555967cc in main ()
```

robintomar1, 2020-06-13:
@tomlogan501  I have been facing the same issue. when i installed the package using 
`apt-get in stall ros-kinetic-teb-local-planner`
the node keeps shutting down the same way you said.
I also tried to build the package from source but was unable to do so and it shows some error with the g2o
```
/navigation_ws/src/navigation/teb_local_planner/src/optimal_planner.cpp:172:91: 
error: no matching function for call to ‘g2o::BlockSolver<g2o::BlockSolverTraits<-1, -1> >::BlockSolver(std::remove_reference<std::unique_ptr<g2o::LinearSolverCSparse<Eigen::Matrix<double, -1, -1> > >&>::type)’
   std::unique_ptr<TEBBlockSolver> block_solver(new TEBBlockSolver(std::move(linear_solver)));
```
tried with both using `kinetic-devel` branch and ` melodic-devel` branch but no luck
ros distro : kinetic
ubuntu: 16.04

tomlogan501, 2020-06-17:
Even in the rviz tutorial with the 3 obstacles ?

robintomar1, 2020-06-17:
> Even in the rviz tutorial with the 3 obstacles ?

yes

wolfv, 2020-08-03:
I am seeing the same issue with packages that I built for melodic. It might be related to g2o?

ljburtz, 2020-09-16:
same issue! after installing from source melodic-devel branch

```
~/utilities/ws_rviz/src/teb_local_planner/cfg$ rviz --display-config ~/software/SpaceRoboticsChallenge/layout/rviz_test_optim.rviz 
[ INFO] [1600266201.514917781]: rviz version 1.13.13
[ INFO] [1600266201.514966501]: compiled against Qt version 5.9.5
[ INFO] [1600266201.514983948]: compiled against OGRE version 1.9.0 (Ghadamon)
[ INFO] [1600266201.518589443]: Forcing OpenGl version 0.
[ INFO] [1600266202.079832253]: Stereo is NOT SUPPORTED
[ INFO] [1600266202.079898053]: OpenGl version: 4.6 (GLSL 4.6).
Segmentation fault (core dumped)

```

radisc, 2020-11-26:
> same issue! after installing from source melodic-devel branch
> 
> when running the rviz and test_optim_node seperately, then only RVIZ will crash, not the test_optim_node
> 
> ```
> ~/utilities/ws_rviz/src/teb_local_planner/cfg$ rviz --display-config ~/software/SpaceRoboticsChallenge/layout/rviz_test_optim.rviz 
> [ INFO] [1600266201.514917781]: rviz version 1.13.13
> [ INFO] [1600266201.514966501]: compiled against Qt version 5.9.5
> [ INFO] [1600266201.514983948]: compiled against OGRE version 1.9.0 (Ghadamon)
> [ INFO] [1600266201.518589443]: Forcing OpenGl version 0.
> [ INFO] [1600266202.079832253]: Stereo is NOT SUPPORTED
> [ INFO] [1600266202.079898053]: OpenGl version: 4.6 (GLSL 4.6).
> Segmentation fault (core dumped)
> ```

I'm having the same issue, i'm using melodic packet manager version and i'm up to date with everything

guannan-he, 2021-06-18:
Similar issue at #131 , deactive "Marker" in display panel and reactive before drag obstacles or publish point might work.