rst-tu-dortmund/teb_local_planner #215: [ROS2] API change incoming

SteveMacenski, 2020-07-08:
Hi,

From https://github.com/ros-planning/navigation2/pull/1857 we're moving the progress and goal checkers internally to the nav2_controller server rather than having to be supplied by the algorithm. As such, we're removing the `isGoalReached` from the controller API that TEB implements. If you need a specialized goal checker plugin, please let us know and we can work to get that included into the base navigation2 stack. If you don't need a specialized one, please look over the default one and make sure that it meets your needs (was used in DWB).

croesmann, 2020-07-19:
Many thanks for letting us know! The current goal checking is very similar to the one from DWB except the changes merged in #43. The check there only addresses TEB with via-points and circular global plans. 

The use case is actually when start and goal are quite similar (w.r.t. position and orientation tolerances),
but the global planner creates a trajectory of considerable lengthnto pass some other waypoints, the current DWB goal checker still returns "goal reached".
Perhaps a more abstract way to integrate this into the goal checker in the base navigation stack would be to include the remaining length of the global plan as another goal condition. This would enable this type of application for all planners.
(The number of via points is derived from the global plan based on a predefined grid, so that the verification of the number of via points or global plan lengths is almost equivalent).
But to be honest, this a use case that does not occur often in applications ;-)

SteveMacenski, 2020-07-20:
Well the number of points can't be relied upon. The sparsity of plans can't be assumed. Some might be very dense, others might not be.

Michael-Equi, 2020-07-26:
Has a decision been reached on this? The change to the nav2 stack prevents the TEB planner form building: `error: ‘bool teb_local_planner::TebLocalPlannerROS::isGoalReached(const PoseStamped&, const Twist&)’ marked ‘override’, but does not override` as of right now.

SteveMacenski, 2020-08-24:
Another API change https://github.com/ros-planning/navigation2/blob/45775b360a8bad5e1fb04cf7fb230a318159df4b/nav2_core/include/nav2_core/controller.hpp we're moving to WeakPtrs over sharedPtrs to deal with cleaner exits in the active state.

eliasdc, 2020-09-01:
> Has a decision been reached on this? The change to the nav2 stack prevents the TEB planner form building: `error: ‘bool teb_local_planner::TebLocalPlannerROS::isGoalReached(const PoseStamped&, const Twist&)’ marked ‘override’, but does not override`.

is there a workaround to fix this?

SteveMacenski, 2020-09-01:
Submit a PR for the master branch updates

closed 2020-10-17 by amakarow