rst-tu-dortmund/teb_local_planner #218: DIfference between two kinds of optim variables arrangement

XiruiLiu038, 2020-07-15:
Hallo Christoph,

I am trying to optimize the computing time of TEB. I found that in the paper **Efficient Optimization using a Sparse Mode** in 2013, the arrangement of optimization variables was `[x0 ... xi ... xn t0 ... ti ... tn]`;  
And in the paper **Kinodynamic Trajectory Optimization and Control for Car-Like Robots** in 2017, the  arrangement became `[x0 t0 ... xi ti ... xn]`
Obviously, the TEB problem results should be the same, but the ordering of the sparse Jacobi matrix **J** and system matrix **H** would be influenced by different arrangements. I was wondering, why two kinds of arrangements have been applied and is there any significant time cost difference during computing **J H** and solving linear equations.

Looking forward to your reply.

croesmann, 2020-07-15:
Hi @XiruiLiu038 ,

well observed ;-)! 
The original papers from 2012 and 2013 use a different cost for time-optimality, i.e. `(t0 + t1 + ...)^2`.
Later, e.g. as described in the paper **Integrated online trajectory planning and optimization in distinctive topologies**,
we switched to `(t0^2 + t1^2 + ...)` which inherently leads to a better structure and more well-posed problem. In the absence of further cost, the solution is actually time-optimal, but in the overall TEB it's more quasi-time-optimal.
The new formulation leads to a banded sparsity structure of the Hessian in case of arrangement [x0 t0 ... xi ti ...].

I cannot remember that I did any benchmarks w.r.t. the old arrangement and the new cost function, however, the banded structure appears to be more natural to me. It might also depend on the block-ordering strategy used in combination with the underlying Cholesky decomposition. 
For the old cost function and the old arrangement, the difference in terms of computation time was quite minimal if I remember correctly.

croesmann, 2020-07-19:
I'll close this for now but still feel free to comment on that question.

closed 2020-07-19 by croesmann