rst-tu-dortmund/teb_local_planner #220: [ROS2] Update teb_params.yaml in order to properly run tb3 simulation

Michael-Equi, 2020-07-25:
OS: Ubuntu 20.04
Dist: ROS2 Foxy Binaries
Nav2: 69977cd83d9040df3422d8a2e564715b6002f3fb
TEB: 8bd805e29ef629a003880e22d67d9529da76a87d

The current teb_params.yaml in the ros2-master branch does not allow the tb3 simulation to successfully run by preventing the controller node from progressing to an active state. 

Steps to replicate:
Build the teb_local_planner package for ROS2 Foxy and run `teb_tb3_simulation_launch.py`.

garyrightbot, 2020-10-20:
I'm facing similar issues. Referring to the sample params, I came up with the following params for the tb3 simulations:

```
amcl:
  ros__parameters:
    use_sim_time: True
    alpha1: 0.2
    alpha2: 0.2
    alpha3: 0.2
    alpha4: 0.2
    alpha5: 0.2
    base_frame_id: "base_footprint"
    beam_skip_distance: 0.5
    beam_skip_error_threshold: 0.9
    beam_skip_threshold: 0.3
    do_beamskip: false
    global_frame_id: "map"
    lambda_short: 0.1
    laser_likelihood_max_dist: 2.0
    laser_max_range: 100.0
    laser_min_range: -1.0
    laser_model_type: "likelihood_field"
    max_beams: 60
    max_particles: 2000
    min_particles: 500
    odom_frame_id: "odom"
    pf_err: 0.05
    pf_z: 0.99
    recovery_alpha_fast: 0.0
    recovery_alpha_slow: 0.0
    resample_interval: 1
    robot_model_type: "differential"
    save_pose_rate: 0.5
    sigma_hit: 0.2
    tf_broadcast: true
    transform_tolerance: 1.0
    update_min_a: 0.2
    update_min_d: 0.25
    z_hit: 0.5
    z_max: 0.05
    z_rand: 0.5
    z_short: 0.05
    scan_topic: scan

amcl_map_client:
  ros__parameters:
    use_sim_time: True

amcl_rclcpp_node:
  ros__parameters:
    use_sim_time: True

bt_navigator:
  ros__parameters:
    use_sim_time: True
    global_frame: map
    robot_base_frame: base_link
    odom_topic: /odom
    default_bt_xml_filename: "navigate_w_replanning_and_recovery.xml"
    plugin_lib_names:
    - nav2_compute_path_to_pose_action_bt_node
    - nav2_follow_path_action_bt_node
    - nav2_back_up_action_bt_node
    - nav2_spin_action_bt_node
    - nav2_wait_action_bt_node
    - nav2_clear_costmap_service_bt_node
    - nav2_is_stuck_condition_bt_node
    - nav2_goal_reached_condition_bt_node
    - nav2_goal_updated_condition_bt_node
    - nav2_initial_pose_received_condition_bt_node
[... 168 lines cut ...]
      plugin: "nav2_recoveries/BackUp"
    wait:
      plugin: "nav2_recoveries/Wait"
    global_frame: odom
    robot_base_frame: base_link
    transform_timeout: 0.1
    use_sim_time: true
    simulate_ahead_time: 2.0
    max_rotational_vel: 1.0
    min_rotational_vel: 0.4
    rotational_acc_lim: 3.2

robot_state_publisher:
  ros__parameters:
    use_sim_time: True
```

However, as soon as I set the goal pose, I constantly keep getting the following error:

```
[controller_server-4] [ERROR] [1603190336.450116718] [controller_server]: No valid trajectories out of 0! 
[controller_server-4] [WARN] [1603190336.450605379] [controller_server_rclcpp_node]: [follow_path] [ActionServer] Aborting handle.
```

I'm unable to tell if this is because of the params or some issues in the ros2 branch.

AlexKaravaev, 2020-11-26:
Any luck on this? Been struggling with params file of teb planner on foxy for a while

SteveMacenski, 2021-03-09:
@doisyg have you tried with the given config file?

PR #280 (2021-03-10): [ros2-master] Updated parameters for ros2 master sync
Resolves #220.
With this params I was able to launch nav2 branch master with teb_local_planner as controller plugin
closed 2021-03-10 by SteveMacenski