rst-tu-dortmund/teb_local_planner #221: Local path following dynamic obstacle near to the goal

mtlazaro, 2020-07-28:
Dear all,

first of all, thanks for your contribution to the robotics community with this local planner.

We are experiencing an strange issue of the TEB local planner where the computed trajectory tries to "follow" a moving obstacle (in this case a person) instead of being recomputed to obtain a shorter path towards the goal.
We observed that this issue appears specially close to the goal. To explain this better, we share here this video:

https://drive.google.com/file/d/13bAjEJ8FIvoU8gy4QJpgZoRpuFeG3td_/view?usp=sharing

In the video you can see the teb plan surrounding the person when the goal is just placed in straight line (green straight line is the global plan). Around 0:43 we give another goal close to the previous one and the local plan is not recomputed. Then, around 0:55 we give a new goal further than the current one and then this time, the local plan is recomputed correctly in straight line.

Ideally we would not like to send different goals to solve this situation and the first "people following" should not happen. I guess this is an issue with the optimizer, that becomes far from the optimal solution and it takes as initial guess the previous deformed one. 

Have you experienced this issue? Could you help in how to tune the TEB so this does not happen or how to detect this situation to "reset" it?

Best,
Mayte.

mxpfeng, 2020-09-09:
I also encountered the same problem. Have you solved this problem？ @mtlazaro

mtlazaro, 2020-09-15:
Hi,
I'm afraid we haven't. 
Maybe one could detect that the differences in length between the computed teb plan and the local or global plan is high. But I don't know how we could later reset the planner since resending the same goal takes no effect.

mxpfeng, 2020-09-17:
> Hi,
> I'm afraid we haven't.
> Maybe one could detect that the differences in length between the computed teb plan and the local or global plan is high. But I don't know how we could later reset the planner since resending the same goal takes no effect.

Thank you for your reply, and I will try to continue to solve this problem. I hope that our new progress can be synchronized here. Thanks