rst-tu-dortmund/teb_local_planner #226: Build failing for teb_local_planner package on ubuntu bionic ros foxy

garyrightbot, 2020-09-09:
```
warry@warry-desktop:~/navigation2_ws$ colcon build --symlink-install
Starting >>> nav2_common
Starting >>> nav_2d_msgs                   
Starting >>> costmap_converter_msgs                                         
Starting >>> nav2_gazebo_spawner                                                
Finished <<< nav2_common [5.64s]                                                                   
Starting >>> nav2_msgs
Starting >>> nav2_voxel_grid                                          
Finished <<< costmap_converter_msgs [7.73s]                             
Starting >>> teb_msgs
Finished <<< nav2_voxel_grid [3.20s]                                     
Finished <<< nav2_gazebo_spawner [8.67s]                                 
Finished <<< nav_2d_msgs [9.74s]                                         
Starting >>> dwb_msgs                                                                        
Finished <<< teb_msgs [6.12s]                                                                       
Finished <<< dwb_msgs [7.30s]                                                                      
Finished <<< nav2_msgs [11.9s]                          
Starting >>> nav2_util 
Finished <<< nav2_util [2.94s]                         
Starting >>> nav_2d_utils
Starting >>> nav2_behavior_tree              
Starting >>> nav2_lifecycle_manager                                                  
Starting >>> nav2_map_server                                                             
Starting >>> nav2_amcl                                                                   
Starting >>> nav2_waypoint_follower                                                      
Finished <<< nav_2d_utils [5.31s]                                                              
Finished <<< nav2_lifecycle_manager [6.42s]                                   
Starting >>> nav2_rviz_plugins
Finished <<< nav2_waypoint_follower [6.36s]                                    
Finished <<< nav2_amcl [7.34s]                                                 
Finished <<< nav2_map_server [7.59s]                                           
Starting >>> nav2_costmap_2d
Finished <<< nav2_rviz_plugins [4.69s]                                           
Finished <<< nav2_behavior_tree [13.2s]                                           
Starting >>> nav2_bt_navigator
Finished <<< nav2_costmap_2d [6.83s]                                                                
Starting >>> nav2_core
Starting >>> costmap_queue                                                                   
Starting >>> costmap_converter                                                                   
Finished <<< nav2_core [1.97s]                                                                      
Starting >>> dwb_core                                                           
Starting >>> nav2_controller                                                    
Starting >>> nav2_navfn_planner                                                 
Finished <<< nav2_bt_navigator [4.30s]                                          
Starting >>> nav2_planner                                              
Finished <<< costmap_queue [4.30s]                                          
Starting >>> nav2_recoveries
Finished <<< costmap_converter [5.18s]                                          
Finished <<< nav2_navfn_planner [3.72s]                               
Finished <<< dwb_core [4.77s]                                          
Starting >>> dwb_critics
Starting >>> dwb_plugins                                                      
Finished <<< nav2_planner [4.51s]                                             
Finished <<< nav2_controller [5.13s]                                          
Finished <<< nav2_recoveries [4.08s]                                                                
Finished <<< dwb_critics [2.90s]                                                                    
Finished <<< dwb_plugins [2.95s]                          
Starting >>> nav2_dwb_controller
Finished <<< nav2_dwb_controller [1.58s]                     
Starting >>> navigation2
Finished <<< navigation2 [1.94s]                     
Starting >>> nav2_bringup
Finished <<< nav2_bringup [1.98s]                     
Starting >>> nav2_system_tests
Starting >>> teb_local_planner                     
Finished <<< nav2_system_tests [3.58s]                                                          
[Processing: teb_local_planner]                                    
[Processing: teb_local_planner]                                        
--- stderr: teb_local_planner                                           
[... 38 lines cut ...]
/home/warry/navigation2_ws/install/nav2_core/include/nav2_core/controller.hpp:74:16: note: 	virtual void nav2_core::Controller::configure(const WeakPtr&, std::__cxx11::string, const std::shared_ptr<tf2_ros::Buffer>&, const std::shared_ptr<nav2_costmap_2d::Costmap2DROS>&)
   virtual void configure(
                ^~~~~~~~~
make[2]: *** [CMakeFiles/teb_local_planner.dir/src/teb_local_planner_ros.cpp.o] Error 1
make[1]: *** [CMakeFiles/teb_local_planner.dir/all] Error 2
make: *** [all] Error 2
---
Failed   <<< teb_local_planner [1min 17s, exited with code 2]
                                        
Summary: 32 packages finished [2min 11s]
  1 package failed: teb_local_planner
  1 package had stderr output: teb_local_planner

```

```
warry@warry-desktop:~/navigation2_ws$ lsb_release -a
No LSB modules are available.
Distributor ID:	Ubuntu
Description:	Ubuntu 18.04.5 LTS
Release:	18.04
Codename:	bionic
```

`Platform: Jetson Xavier NX`

garyrightbot, 2020-09-10:
Seems related to https://github.com/rst-tu-dortmund/teb_local_planner/issues/215

amakarow, 2020-10-17:
Closed because #233 was merged.

closed 2020-10-17 by amakarow

xivien, 2020-10-23:
After installing TEB on latest master branch (ROS2 Foxy, ubuntu 20.04.1LTS) I still have errors, many of which are the same as in the issue:
Installation steps: 
```bash
sudo apt-get install ros-foxy-libg2o
git clone --branch ros2-master https://github.com/rst-tu-dortmund/teb_local_planner.git
git clone --branch ros2 https://github.com/rst-tu-dortmund/costmap_converter.git
```

<details>
  <summary>Build Log</summary>

```bash
--- stderr: teb_local_planner                               
In file included from /home/xivien/ros2_ws/src/teb_local_planner/teb_local_planner/src/teb_local_planner_ros.cpp:39:
/home/xivien/ros2_ws/src/teb_local_planner/teb_local_planner/include/teb_local_planner/teb_local_planner_ros.h:108:8: error: ‘void teb_local_planner::TebLocalPlannerROS::configure(const WeakPtr&, std::string, const std::shared_ptr<tf2_ros::Buffer>&, const std::shared_ptr<nav2_costmap_2d::Costmap2DROS>&)’ marked ‘override’, but does not override
  108 |   void configure(
      |        ^~~~~~~~~
/home/xivien/ros2_ws/src/teb_local_planner/teb_local_planner/src/teb_local_planner_ros.cpp: In constructor ‘teb_local_planner::TebLocalPlannerROS::TebLocalPlannerROS()’:
/home/xivien/ros2_ws/src/teb_local_planner/teb_local_planner/src/teb_local_planner_ros.cpp:72:101: error: no matching function for call to ‘std::weak_ptr<rclcpp_lifecycle::LifecycleNode>::weak_ptr(std::nullptr_t)’
   72 |           last_preferred_rotdir_(RotType::none), initialized_(false)
      |                                                                    ^

In file included from /usr/include/c++/9/memory:81,
                 from /opt/ros/foxy/include/pluginlib/class_loader.hpp:51,
                 from /home/xivien/ros2_ws/src/teb_local_planner/teb_local_planner/include/teb_local_planner/teb_local_planner_ros.h:42,
                 from /home/xivien/ros2_ws/src/teb_local_planner/teb_local_planner/src/teb_local_planner_ros.cpp:39:
/usr/include/c++/9/bits/shared_ptr.h:560:2: note: candidate: ‘template<class _Yp, class> std::weak_ptr<_Tp>::weak_ptr(std::weak_ptr<_Yp>&&)’
  560 |  weak_ptr(weak_ptr<_Yp>&& __r) noexcept
      |  ^~~~~~~~
/usr/include/c++/9/bits/shared_ptr.h:560:2: note:   template argument deduction/substitution failed:
/home/xivien/ros2_ws/src/teb_local_planner/teb_local_planner/src/teb_local_planner_ros.cpp:72:101: note:   mismatched types ‘std::weak_ptr<_Tp>’ and ‘std::nullptr_t’
   72 |           last_preferred_rotdir_(RotType::none), initialized_(false)
      |                                                                    ^

In file included from /usr/include/c++/9/memory:81,
                 from /opt/ros/foxy/include/pluginlib/class_loader.hpp:51,
                 from /home/xivien/ros2_ws/src/teb_local_planner/teb_local_planner/include/teb_local_planner/teb_local_planner_ros.h:42,
                 from /home/xivien/ros2_ws/src/teb_local_planner/teb_local_planner/src/teb_local_planner_ros.cpp:39:
/usr/include/c++/9/bits/shared_ptr.h:557:7: note: candidate: ‘std::weak_ptr<_Tp>::weak_ptr(std::weak_ptr<_Tp>&&) [with _Tp = rclcpp_lifecycle::LifecycleNode]’
  557 |       weak_ptr(weak_ptr&&) noexcept = default;
      |       ^~~~~~~~
/usr/include/c++/9/bits/shared_ptr.h:557:16: note:   no known conversion for argument 1 from ‘std::nullptr_t’ to ‘std::weak_ptr<rclcpp_lifecycle::LifecycleNode>&&’
  557 |       weak_ptr(weak_ptr&&) noexcept = default;
      |                ^~~~~~~~~~
/usr/include/c++/9/bits/shared_ptr.h:554:2: note: candidate: ‘template<class _Yp, class> std::weak_ptr<_Tp>::weak_ptr(const std::weak_ptr<_Yp>&)’
  554 |  weak_ptr(const weak_ptr<_Yp>& __r) noexcept
      |  ^~~~~~~~
/usr/include/c++/9/bits/shared_ptr.h:554:2: note:   template argument deduction/substitution failed:
/home/xivien/ros2_ws/src/teb_local_planner/teb_local_planner/src/teb_local_planner_ros.cpp:72:101: note:   mismatched types ‘const std::weak_ptr<_Tp>’ and ‘std::nullptr_t’
   72 |           last_preferred_rotdir_(RotType::none), initialized_(false)
      |                                                                    ^

In file included from /usr/include/c++/9/memory:81,
                 from /opt/ros/foxy/include/pluginlib/class_loader.hpp:51,
                 from /home/xivien/ros2_ws/src/teb_local_planner/teb_local_planner/include/teb_local_planner/teb_local_planner_ros.h:42,
                 from /home/xivien/ros2_ws/src/teb_local_planner/teb_local_planner/src/teb_local_planner_ros.cpp:39:
/usr/include/c++/9/bits/shared_ptr.h:551:7: note: candidate: ‘std::weak_ptr<_Tp>::weak_ptr(const std::weak_ptr<_Tp>&) [with _Tp = rclcpp_lifecycle::LifecycleNode]’
  551 |       weak_ptr(const weak_ptr&) noexcept = default;
      |       ^~~~~~~~
/usr/include/c++/9/bits/shared_ptr.h:551:16: note:   no known conversion for argument 1 from ‘std::nullptr_t’ to ‘const std::weak_ptr<rclcpp_lifecycle::LifecycleNode>&’
  551 |       weak_ptr(const weak_ptr&) noexcept = default;
      |                ^~~~~~~~~~~~~~~
/usr/include/c++/9/bits/shared_ptr.h:548:2: note: candidate: ‘template<class _Yp, class> std::weak_ptr<_Tp>::weak_ptr(const std::shared_ptr<_Yp>&)’
  548 |  weak_ptr(const shared_ptr<_Yp>& __r) noexcept
      |  ^~~~~~~~
/usr/include/c++/9/bits/shared_ptr.h:548:2: note:   template argument deduction/substitution failed:
/home/xivien/ros2_ws/src/teb_local_planner/teb_local_planner/src/teb_local_planner_ros.cpp:72:101: note:   mismatched types ‘const std::shared_ptr<_Tp>’ and ‘std::nullptr_t’
[... cut ...]
In file included from /home/xivien/ros2_ws/src/teb_local_planner/teb_local_planner/include/teb_local_planner/teb_local_planner_ros.h:47,
                 from /home/xivien/ros2_ws/src/teb_local_planner/teb_local_planner/src/teb_local_planner_ros.cpp:39:
/opt/ros/foxy/include/nav2_core/controller.hpp:74:16: note: 	‘virtual void nav2_core::Controller::configure(const SharedPtr&, std::string, const std::shared_ptr<tf2_ros::Buffer>&, const std::shared_ptr<nav2_costmap_2d::Costmap2DROS>&)’
   74 |   virtual void configure(
      |                ^~~~~~~~~

```

</details>

SteveMacenski, 2020-10-25:
@amakarow those changes are for navigation2 mainline targetting ros2 master, not specifically foxy's release. This ticket should be reopened. A new `foxy-devel` branch should be created that can have a buildable foxy version of TEB.

amakarow, 2020-10-25:
Done. See #239.