rst-tu-dortmund/teb_local_planner #239: [ROS2] foxy-devel branch + foxy compatibility

SteveMacenski, 2020-10-25:
Currently there's a ros2 main branch which works, as well as a dashing and eloquent branch. But there is no `foxy` branch and as the latest LTS distribution that people are using, we need to have foxy supported. 

I suggest making a foxy branch off of `ros2` and rolling back some of the `ros2` changes for foxy compatibility.

amakarow, 2020-10-25:
I created foxy-devel branch and dropped the commits introduced by your PR #233. I have not tested it yet.

closed 2020-12-24 by SteveMacenski

eliasdc, 2021-01-04:
build does not succeed on Foxy for me: 

```
--- stderr: teb_local_planner                                 
In file included from /ros_ws/src/plugins/controllers/teb_local_planner/teb_local_planner/src/teb_local_planner_ros.cpp:39:
/ros_ws/src/plugins/controllers/teb_local_planner/teb_local_planner/include/teb_local_planner/teb_local_planner_ros.h:139:8: error: ‘bool teb_local_planner::TebLocalPlannerROS::isGoalReached(const PoseStamped&, const Twist&)’ marked ‘override’, but does not override
  139 |   bool isGoalReached(
      |        ^~~~~~~~~~~~~
make[2]: *** [CMakeFiles/teb_local_planner.dir/build.make:174: CMakeFiles/teb_local_planner.dir/src/teb_local_planner_ros.cpp.o] Error 1
make[1]: *** [CMakeFiles/Makefile2:98: CMakeFiles/teb_local_planner.dir/all] Error 2
make: *** [Makefile:161: all] Error 2
---
Failed   <<< teb_local_planner [14.5s, exited with code 2]
```