rst-tu-dortmund/teb_local_planner #240: [ROS2] Sync ROS2 branch with PRs for ROS1

SteveMacenski, 2020-10-26:
Analog to https://github.com/rst-tu-dortmund/teb_local_planner/pull/235 where we need to keep ROS2 up to date with these changes too - since ROS1 will be soon completely deprecated and ROS2 will be all what's left.

SteveMacenski, 2020-10-26:
Especially the loop fix and any substantive bug / performance boosts (like #234 when merged)

eliasdc, 2021-02-11:
Can we expect a sync? I think a lot of features are really necessary in the Foxy version. #231 is one of those features we really need.

SteveMacenski, 2021-03-08:
@eliasdc PRs are appreciated

SteveMacenski, 2021-03-08:
https://github.com/rst-tu-dortmund/teb_local_planner/pull/198 another PR to forward port to ros2-master

SteveMacenski, 2021-03-11:
https://github.com/rst-tu-dortmund/teb_local_planner/issues/283 supersedes

closed 2021-03-11 by SteveMacenski