rst-tu-dortmund/teb_local_planner #242: rotation cost (what decides if the robot turns left or right when goal is in-place)

greymfm, 2020-11-02:
Hello,

First of all : congrats to all of the RST TU Dortmund team - you are doing a great job!  :-)
 
My differential robot (front differential steering) can turn in-place. Sometimes 'teb_local_planner' chooses the most efficient side to turn in-place to the goal pose, sometimes not. 

Example: Turning in-place 90 degree would be closer to the goal pose than turning (360-90) degree - but at certain locations (always the same locations) it chooses the less-efficient turning side. 

How could I optimize this? The same happens with the differential robot from the 'teb_local_planner' tutorials. 

Here's a demo: https://drive.google.com/file/d/1GHitcXlAKp3CfvwvEsnKg1-tFLz5RkJM/view
 
Thanks for any tips :-)
Regards,
Alexander

jcmonteiro, 2020-12-11:
If I am not mistaken, this will depend heavily on the initial pose graph. You could try to enable homotopy to look for different topologies and see if that changes things.

ZhangJunfeng199, 2023-11-16:
I encountered the same problem. How did you solve it? Looking forward to your reply.