rst-tu-dortmund/teb_local_planner #252: TEB Steering output oscillate

droter, 2020-11-17:
I was plotting out the cmd_vel.angular.z message from the TEB planner.  How do I prevent the steering from oscillating?  This happens both as rotational velocity and steer angle setting in TEB.

[image]

jediofgever, 2020-11-20:
I might observed a similar behavior on my Ackermann drive robot. Though haven’t plot the value from cmd_vel as I thought the oscillation might be caused because of the poor tuned PID values of steering control. This could well be the same issue,

LotfiZ, 2020-12-10:
I have the same issue with translation velocities, it oscillate between 0 and the max vel that i set, i used the same setup in simulation and it was not the case.
[image]

jcmonteiro, 2020-12-11:
These are usually examples of poor tuning. What set of parameters are you using?

tianb03, 2021-04-17:
I also have a similar situation recently when I test with a mars rover with Ackermann drive. The cmd_vel oscillation should not be an issue because when the robot's moving direction changes, the angular z will be change sign directly.

However, when the robot makes an "r" movement, the Ackermann steering seems will not take the  acc_lim_theta into consideration, maybe because there is also very little translational velocity. Then if the robot needs to move back and forth several times to adjust its pose, the steering wheel will oscillate dramatically. 

Currently, I am not sure how to solve this problem. In stage simulation, it seems smooth just because there is no real mechanism. It is really appreciated that someone could provide a patch for this issue.

jachyshu, 2024-06-10:
it seemed that i met the same situation, wondering if u solved it.