rst-tu-dortmund/teb_local_planner #257: when robot navigates close to wall or obstacle, planner publish large amout of teb_poses

charon-cheung, 2020-12-03:
I upload the [video](https://www.youtube.com/watch?v=ETORdzBcT0w), this rarely happens, I found the global path changed in the end, is that the reason ?

jcmonteiro, 2020-12-11:
It could be that this is caused by the global path changing to an unfeasible final pose. The first and the final poses are fixed vertices in the optimization.

It looks like the algorithm had converged well from your video, but at the very end, your final pose changed to an unfeasible one. You could try to reproduce in two ways:

- Change the final pose at the very end to an unfeasible one.
- Change the final pose at the very end to another feasible one.

Both of the above roughly at the same final position (just like in your video).

charon-cheung, 2020-12-15:
> It could be that this is caused by the global path changing to an unfeasible final pose. The first and the final poses are fixed vertices in the optimization.
>
> It looks like the algorithm had [...]

thanks, I will try to reproduce as you say

Noah-Lau, 2023-09-05:
I got a similar problem, have you figure it out?