rst-tu-dortmund/teb_local_planner #264: Error when launch test_optim_node.launch

GettingChen, 2020-12-20:
When I run :
`
roslaunch teb_local_planner test_optim_node.launch
`
I get error:
`
ERROR: cannot launch node of type [teb_local_planner/test_optim_node]: can't locate node [test_optim_node] in package [teb_local_planner]

`
I feel confused ,maybe there is something wrong with tf

[image]
Can anyone help me?Thanks in advance.

GettingChen, 2020-12-21:
Now it works,but when i run
`rosrun teb_local_planner visualize_velocity_profile.py`

I got some errors
`Traceback (most recent call last): File "/home/cjt/turtlebot_ws/src/teb_local_planner_tutorials/scripts/visualize_velocity_profile.py", line 9, in <module> from teb_local_planner.msg import FeedbackMsg, TrajectoryMsg, TrajectoryPointMsg ImportError: No module named teb_local_planner.msg`

It is not supposed to happen.Can anyone tell me how to solve this?

gycyr, 2020-12-27:
I think the first problem you can take a closer look at the tutorial can be solved.http://wiki.ros.org/teb_local_planner/Tutorials/Inspect%20optimization%20feedback
The second problem I have not encountered, but I think you should check whether you have the complete teb package