rst-tu-dortmund/teb_local_planner #266: Which version of the navigation package does the lasted "ros2-master" branch was tested with?

ByLiZhao, 2020-12-23:
Which version of the navigation package does the lasted "ros2-master" branch was tested with?

While building "teb_local_planner", ros2-master branch with the latest version the navigation2 project, foxy-devel branch, the compiler complains that: 

> In file included from /home/johnzli/ros2_ws/src/teb_local_planner/teb_local_planner/src/teb_local_planner_ros.cpp:39:
/home/johnzli/ros2_ws/src/teb_local_planner/teb_local_planner/include/teb_local_planner/teb_local_planner_ros.h:108:8: error: ‘void teb_local_planner::TebLocalPlannerROS::configure(const WeakPtr&, std::string, const std::shared_ptr<tf2_ros::Buffer>&, const std::shared_ptr<nav2_costmap_2d::Costmap2DROS>&)’ marked ‘override’, but does not override
  108 |   void configure(
      |        ^~~~~~~~~
/home/johnzli/ros2_ws/src/teb_local_planner/teb_local_planner/src/teb_local_planner_ros.cpp: In constructor ‘teb_local_planner::TebLocalPlannerROS::TebLocalPlannerROS()’:
/home/johnzli/ros2_ws/src/teb_local_planner/teb_local_planner/src/teb_local_planner_ros.cpp:72:101: error: no matching function for call to ‘std::weak_ptr<rclcpp_lifecycle::LifecycleNode>::weak_ptr(std::nullptr_t)’
   72 |                                            last_preferred_rotdir_(RotType::none), initialized_(false)
      |                                                                                                     ^
In file included from /usr/include/c++/9/memory:81,
                 from /opt/ros/foxy/include/pluginlib/class_loader.hpp:51,
                 from /home/johnzli/ros2_ws/src/teb_local_planner/teb_local_planner/include/teb_local_planner/teb_local_planner_ros.h:42,
                 from /home/johnzli/ros2_ws/src/teb_local_planner/teb_local_planner/src/teb_local_planner_ros.cpp:39:
/usr/include/c++/9/bits/shared_ptr.h:560:2: note: candidate: ‘template<class _Yp, class> std::weak_ptr<_Tp>::weak_ptr(std::weak_ptr<_Yp>&&)’
  560 |  weak_ptr(weak_ptr<_Yp>&& __r) noexcept
      |  ^~~~~~~~
/usr/include/c++/9/bits/shared_ptr.h:560:2: note:   template argument deduction/substitution failed:
/home/johnzli/ros2_ws/src/teb_local_planner/teb_local_planner/src/teb_local_planner_ros.cpp:72:101: note:   mismatched types ‘std::weak_ptr<_Tp>’ and ‘std::nullptr_t’
   72 |                                            last_preferred_rotdir_(RotType::none), initialized_(false)
      |                                                                                                     ^
In file included from /usr/include/c++/9/memory:81,
                 from /opt/ros/foxy/include/pluginlib/class_loader.hpp:51,
                 from /home/johnzli/ros2_ws/src/teb_local_planner/teb_local_planner/include/teb_local_planner/teb_local_planner_ros.h:42,
                 from /home/johnzli/ros2_ws/src/teb_local_planner/teb_local_planner/src/teb_local_planner_ros.cpp:39:
/usr/include/c++/9/bits/shared_ptr.h:557:7: note: candidate: ‘std::weak_ptr<_Tp>::weak_ptr(std::weak_ptr<_Tp>&&) [with _Tp = rclcpp_lifecycle::LifecycleNode]’
  557 |       weak_ptr(weak_ptr&&) noexcept = default;
      |       ^~~~~~~~
/usr/include/c++/9/bits/shared_ptr.h:557:16: note:   no known conversion for argument 1 from ‘std::nullptr_t’ to ‘std::weak_ptr<rclcpp_lifecycle::LifecycleNode>&&’
  557 |       weak_ptr(weak_ptr&&) noexcept = default;
      |                ^~~~~~~~~~
/usr/include/c++/9/bits/shared_ptr.h:554:2: note: candidate: ‘template<class _Yp, class> std::weak_ptr<_Tp>::weak_ptr(const std::weak_ptr<_Yp>&)’
  554 |  weak_ptr(const weak_ptr<_Yp>& __r) noexcept
      |  ^~~~~~~~
/usr/include/c++/9/bits/shared_ptr.h:554:2: note:   template argument deduction/substitution failed:
/home/johnzli/ros2_ws/src/teb_local_planner/teb_local_planner/src/teb_local_planner_ros.cpp:72:101: note:   mismatched types ‘const std::weak_ptr<_Tp>’ and ‘std::nullptr_t’
   72 |                                            last_preferred_rotdir_(RotType::none), initialized_(false)
      |                                                                                                     ^
In file included from /usr/include/c++/9/memory:81,
                 from /opt/ros/foxy/include/pluginlib/class_loader.hpp:51,
                 from /home/johnzli/ros2_ws/src/teb_local_planner/teb_local_planner/include/teb_local_planner/teb_local_planner_ros.h:42,
                 from /home/johnzli/ros2_ws/src/teb_local_planner/teb_local_planner/src/teb_local_planner_ros.cpp:39:
/usr/include/c++/9/bits/shared_ptr.h:551:7: note: candidate: ‘std::weak_ptr<_Tp>::weak_ptr(const std::weak_ptr<_Tp>&) [with _Tp = rclcpp_lifecycle::LifecycleNode]’
  551 |       weak_ptr(const weak_ptr&) noexcept = default;
      |       ^~~~~~~~
/usr/include/c++/9/bits/shared_ptr.h:551:16: note:   no known conversion for argument 1 from ‘std::nullptr_t’ to ‘const std::weak_ptr<rclcpp_lifecycle::LifecycleNode>&’
  551 |       weak_ptr(const weak_ptr&) noexcept = default;
      |                ^~~~~~~~~~~~~~~
/usr/include/c++/9/bits/shared_ptr.h:548:2: note: candidate: ‘template<class _Yp, class> std::weak_ptr<_Tp>::weak_ptr(const std::shared_ptr<_Yp>&)’
  548 |  weak_ptr(const shared_ptr<_Yp>& __r) noexcept
      |  ^~~~~~~~
/usr/include/c++/9/bits/shared_ptr.h:548:2: note:   template argument deduction/substitution failed:
/home/johnzli/ros2_ws/src/teb_local_planner/teb_local_planner/src/teb_local_planner_ros.cpp:72:101: note:   mismatched types ‘const std::shared_ptr<_Tp>’ and ‘std::nullptr_t’
   72 |                                            last_preferred_rotdir_(RotType::none), initialized_(false)
      |                                                                                                     ^
In file included from /usr/include/c++/9/memory:81,
                 from /opt/ros/foxy/include/pluginlib/class_loader.hpp:51,
                 from /home/johnzli/ros2_ws/src/teb_local_planner/teb_local_planner/include/teb_local_planner/teb_local_planner_ros.h:42,
                 from /home/johnzli/ros2_ws/src/teb_local_planner/teb_local_planner/src/teb_local_planner_ros.cpp:39:
/usr/include/c++/9/bits/shared_ptr.h:544:17: note: candidate: ‘constexpr std::weak_ptr<_Tp>::weak_ptr() [with _Tp = rclcpp_lifecycle::LifecycleNode]’
  544 |       constexpr weak_ptr() noexcept = default;
      |                 ^~~~~~~~
/usr/include/c++/9/bits/shared_ptr.h:544:17: note:   candidate expects 0 arguments, 1 provided
/home/johnzli/ros2_ws/src/teb_local_planner/teb_local_planner/src/teb_local_planner_ros.cpp:72:101: error: ‘rclcpp::Logger::Logger()’ is private within this context
   72 |                                            last_preferred_rotdir_(RotType::none), initialized_(false)
      |                                                                                                     ^
In file included from /opt/ros/foxy/include/rclcpp/logging.hpp:23,
                 from /opt/ros/foxy/include/rclcpp/service.hpp:34,
[... 12 lines cut ...]
   88 |   Logger()
      |   ^~~~~~
/home/johnzli/ros2_ws/src/teb_local_planner/teb_local_planner/src/teb_local_planner_ros.cpp: In member function ‘void teb_local_planner::TebLocalPlannerROS::configure(const WeakPtr&, std::string, const std::shared_ptr<tf2_ros::Buffer>&, const std::shared_ptr<nav2_costmap_2d::Costmap2DROS>&)’:
/home/johnzli/ros2_ws/src/teb_local_planner/teb_local_planner/src/teb_local_planner_ros.cpp:210:8: error: declaration of ‘auto node’ shadows a parameter
  210 |   auto node = nh_.lock();
      |        ^~~~
In file included from /opt/ros/foxy/include/class_loader/class_loader_core.hpp:57,
                 from /opt/ros/foxy/include/class_loader/class_loader.hpp:55,
                 from /opt/ros/foxy/include/class_loader/multi_library_class_loader.hpp:52,
                 from /opt/ros/foxy/include/pluginlib/class_loader.hpp:58,
                 from /home/johnzli/ros2_ws/src/teb_local_planner/teb_local_planner/include/teb_local_planner/teb_local_planner_ros.h:42,
                 from /home/johnzli/ros2_ws/src/teb_local_planner/teb_local_planner/src/teb_local_planner_ros.cpp:39:
/opt/ros/foxy/include/class_loader/meta_object.hpp: In instantiation of ‘B* class_loader::impl::MetaObject<C, B>::create() const [with C = teb_local_planner::TebLocalPlannerROS; B = nav2_core::Controller]’:
/opt/ros/foxy/include/class_loader/meta_object.hpp:190:7:   required from here
/opt/ros/foxy/include/class_loader/meta_object.hpp:192:12: error: invalid new-expression of abstract class type ‘teb_local_planner::TebLocalPlannerROS’
  192 |     return new C;
[... cut ...]
In file included from /home/johnzli/ros2_ws/src/teb_local_planner/teb_local_planner/include/teb_local_planner/teb_local_planner_ros.h:47,
                 from /home/johnzli/ros2_ws/src/teb_local_planner/teb_local_planner/src/teb_local_planner_ros.cpp:39:
/home/johnzli/ros2_ws/install/nav2_core/include/nav2_core/controller.hpp:74:16: note: 	‘virtual void nav2_core::Controller::configure(const SharedPtr&, std::string, const std::shared_ptr<tf2_ros::Buffer>&, const std::shared_ptr<nav2_costmap_2d::Costmap2DROS>&)’
   74 |   virtual void configure(

SteveMacenski, 2020-12-24:
PRs to make TEB work with the new API changes would be appreciated. We recently changed the function signatures and TEB hasn’t been updated yet for it.

SteveMacenski, 2021-03-09:
`main` is the branch `ros2-master` supports. Has been updated to build with nav2 again.

closed 2021-03-09 by SteveMacenski