rst-tu-dortmund/teb_local_planner #269: use teb local planner in non 2D Map ?

jediofgever, 2021-01-13:
Hi @croesmann ,
I am interested to know what architectural changes would be required to make `teb_local_planner` to control a robot with a non 2D occupancy grid(costmap) ? 
I highly appreciate your feedback on this.

jcmonteiro, 2021-01-13:
Does the robot move in 2D space?

jediofgever, 2021-01-13:
The robot moves in 3D space. However controlled states are still 3(x,y,yaw)
it is more related to environment representation, and avoiding obstacles in this representation.

jcmonteiro, 2021-01-13:
Why not cast your 3D data onto 2D?

jediofgever, 2021-01-13:
That’s the initial option but with casting 2D I will not be able to represent elevation , ramps,  bumps that aren’t necessarily obstacle

torgeros, 2021-01-14:
> 
> 
> Why not cast your 3D data onto 2D?

The problem is that odometry in relation to the map is not reliable if the floor is not plain (moving distance is not equal to distance on map)

jediofgever, 2021-01-14:
That will highly depend on the source of your odometetry, some of visual/lidar based SLAM will provide accuracy enough odometry in my opinion

torgeros, 2021-01-14:
it is no longer odometry then but yeah.. but in our usecase for example we only have odometry for guessing our own position and that does indeed not work _that well_ on ramps...

jediofgever, 2021-01-15:
do you have some other mechanism to perform absolute localization within a map or you just rely on odometery ? 
and do you mind directing me link of your project if its open sourced

jcmonteiro, 2021-01-15:
I don't know your constraints, but I would say that if you are relying solely on odometry then you need to tackle that first. Then, you could most likely cast your data to 2D and you TEB as usual.

torgeros, 2021-01-15:
this is not about me, we have a 2D map and tackeled our problems.
sadly we are participating in a competition, so our source code is not open source

closed 2021-01-23 by jediofgever