rst-tu-dortmund/teb_local_planner #271: How to determine the origin point of footprint while applying the teb_local_planner on a car-like robot?

FishInWave, 2021-01-25:
Thanks for your outstanding work.
I have read [teb_local_plannerTutorialsObstacle Avoidance and Robot Footprint Model](http://wiki.ros.org/teb_local_planner/Tutorials/Obstacle%20Avoidance%20and%20Robot%20Footprint%20Model) to figure out how to determine the  origin point of footprint while applying the` teb_local_planner` on a car-like robot.
 But I still cannot understand what this sentence means: 

> The robots axis of rotation is assumed to be located at [0,0] (Unit: meters).

Assume there is a car-like robot with an Ackerman chassis, whose outer contour is 1 m* 0.8m, and I'd like to use teb as its local planner.
1. Should his parameters in `teb_local_planner_params.yaml` be like as follow?
```
 footprint_model: # types: "point", "circular", "two_circles", "line", "polygon" 
  type: "polygon"
  vertices: [[-0.5, -0.4], [-0.5, 0.4],[0.5,0.4],[0.5, -0.4]]
```
2. should the `footprint` in `costmap_common_params.yaml` be the same as above?

Additionally, I noted the `footprint` of [`https://github.com/rst-tu-dortmund/teb_local_planner_tutorials/blob/melodic-devel/cfg/carlike/costmap_common_params.yaml`]( https://github.com/rst-tu-dortmund/teb_local_planner_tutorials/blob/melodic-devel/cfg/carlike/costmap_common_params.yaml) is:
`footprint: [ [-0.1,-0.125], [0.5,-0.125], [0.5,0.125], [-0.1,0.125] ]`.
Why the origin of footprint is to the left?

laszloturanyi, 2021-02-22:
Hi,
[teb_local_plannerTutorialsObstacle Avoidance and Robot Footprint Model](http://wiki.ros.org/teb_local_planner/Tutorials/Obstacle%20Avoidance%20and%20Robot%20Footprint%20Model)

It suggests:
**Important**: For car-like robots, the pose [0,0] is located at the rear-axle (axis of rotation)!

Hence, your origin should be more to the rear, for sure.

FishInWave, 2021-02-26:
> rear

Hi, I want to reconfirm : "located at the axle" means that the pose [0,0] is located at the center of the rear-axle? i.e.  
car contour coordinates are like 
```
 footprint_model: # types: "point", "circular", "two_circles", "line", "polygon" 
  type: "polygon"
  vertices: [[-0.5, -0.4], [-0.5, 0.4],[0.5,0.4],[0.5, -0.4]]
```