rst-tu-dortmund/teb_local_planner #273: [TEB Local Planner] Goal is aborting for ROS2 Foxy

sachinkum0009, 2021-02-09:
TEB local planner is able to find the trajectory but the robot is not moving. It rotates a little and then abort the goal

```bash
##################################################################
				Log File for Teb Local Planner
##################################################################

[controller_server-4] [ERROR] [1612853914.973906030] [controller_server]: Extrapolation Error: Lookup would require extrapolation into the future.  Requested time 1612853914.462717 but the latest data is at time 69.486000, when looking up transform from frame [map] to frame [odom]
[controller_server-4] 
[controller_server-4] [ERROR] [1612853914.974122751] [controller_server]: Global Frame: odom Plan Frame size 69: map
[controller_server-4] 
[controller_server-4] [ERROR] [1612853914.974198322] [controller_server]: Could not transform the global plan to the frame of the controller
[controller_server-4] [WARN] [1612853914.974529978] [controller_server_rclcpp_node]: [follow_path] [ActionServer] Aborting handle.
[controller_server-4] [INFO] [1612853914.974612436] [controller_server]: Received a goal, begin computing control effort.
[controller_server-4] [ERROR] [1612853915.485176247] [controller_server]: Extrapolation Error: Lookup would require extrapolation into the future.  Requested time 1612853914.462717 but the latest data is at time 69.682000, when looking up transform from frame [map] to frame [odom]
[controller_server-4] 
[controller_server-4] [ERROR] [1612853915.485440914] [controller_server]: Global Frame: odom Plan Frame size 69: map
[controller_server-4] 
[controller_server-4] [ERROR] [1612853915.485606161] [controller_server]: Could not transform the global plan to the frame of the controller
[controller_server-4] [WARN] [1612853915.486009813] [controller_server_rclcpp_node]: [follow_path] [ActionServer] Aborting handle.
[controller_server-4] [INFO] [1612853915.511635978] [controller_server]: Received a goal, begin computing control effort.
[controller_server-4] [INFO] [1612853915.519049492] [local_costmap.local_costmap]: Received request to clear entirely the local_costmap
[planner_server-5] [INFO] [1612853915.523121295] [global_costmap.global_costmap]: Received request to clear entirely the global_costmap
[recoveries_server-6] [INFO] [1612853915.534351158] [recoveries_server]: Attempting spin
[recoveries_server-6] [INFO] [1612853915.534435933] [recoveries_server]: Turning 1.57 for spin recovery.
[controller_server-4] [ERROR] [1612853916.017842535] [controller_server]: Extrapolation Error: Lookup would require extrapolation into the future.  Requested time 1612853915.471823 but the latest data is at time 69.884000, when looking up transform from frame [map] to frame [odom]
[controller_server-4] 
[controller_server-4] [ERROR] [1612853916.017915436] [controller_server]: Global Frame: odom Plan Frame size 69: map
[controller_server-4] 
[controller_server-4] [ERROR] [1612853916.017954136] [controller_server]: Could not transform the global plan to the frame of the controller
[controller_server-4] [WARN] [1612853916.018051706] [controller_server_rclcpp_node]: [follow_path] [ActionServer] Aborting handle.
[recoveries_server-6] [INFO] [1612853916.535182490] [recoveries_server]: spin running...
[recoveries_server-6] [INFO] [1612853917.534517168] [recoveries_server]: spin running...
[recoveries_server-6] [INFO] [1612853918.534640689] [recoveries_server]: spin running...
[recoveries_server-6] [INFO] [1612853919.534443093] [recoveries_server]: spin running...
[recoveries_server-6] [INFO] [1612853919.839228088] [recoveries_server]: spin completed successfully
[recoveries_server-6] [INFO] [1612853919.866411043] [recoveries_server]: Attempting wait
[recoveries_server-6] [INFO] [1612853920.868645751] [recoveries_server]: wait running...
[recoveries_server-6] [INFO] [1612853921.868524167] [recoveries_server]: wait running...
[recoveries_server-6] [INFO] [1612853922.868519721] [recoveries_server]: wait running...
[recoveries_server-6] [INFO] [1612853923.868571869] [recoveries_server]: wait running...
[recoveries_server-6] [INFO] [1612853924.868453029] [recoveries_server]: wait running...
[recoveries_server-6] [INFO] [1612853924.870746763] [recoveries_server]: wait completed successfully
[controller_server-4] [INFO] [1612853924.925362466] [controller_server]: Received a goal, begin computing control effort.
[controller_server-4] [ERROR] [1612853925.430621036] [controller_server]: Extrapolation Error: Lookup would require extrapolation into the future.  Requested time 1612853924.901750 but the latest data is at time 73.688000, when looking up transform from frame [map] to frame [odom]
[controller_server-4] 
[controller_server-4] [ERROR] [1612853925.430922663] [controller_server]: Global Frame: odom Plan Frame size 68: map
[controller_server-4] 
[controller_server-4] [ERROR] [1612853925.431111835] [controller_server]: Could not transform the global plan to the frame of the controller
[controller_server-4] [WARN] [1612853925.431575689] [controller_server_rclcpp_node]: [follow_path] [ActionServer] Aborting handle.
[controller_server-4] [INFO] [1612853925.432280776] [controller_server]: Received a goal, begin computing control effort.
[controller_server-4] [ERROR] [1612853925.939998539] [controller_server]: Extrapolation Error: Lookup would require extrapolation into the future.  Requested time 1612853924.901750 but the latest data is at time 73.895000, when looking up transform from frame [map] to frame [odom]
[controller_server-4] 
[controller_server-4] [ERROR] [1612853925.940128851] [controller_server]: Global Frame: odom Plan Frame size 68: map
[controller_server-4] 
[controller_server-4] [ERROR] [1612853925.940192236] [controller_server]: Could not transform the global plan to the frame of the controller
[controller_server-4] [WARN] [1612853925.940350284] [controller_server_rclcpp_node]: [follow_path] [ActionServer] Aborting handle.
[planner_server-5] [INFO] [1612853925.957458893] [planner_server]: Goal was canceled. Canceling planning action.
[planner_server-5] [WARN] [1612853925.958105452] [planner_server_rclcpp_node]: [compute_path_to_pose] [ActionServer] Client requested to cancel the goal. Cancelling.
[controller_server-4] [INFO] [1612853925.966592728] [local_costmap.local_costmap]: Received request to clear entirely the local_costmap
[bt_navigator-7] [WARN] [1612853925.970601835] [bt_navigator_rclcpp_node]: Node timed out while executing service call to local_costmap/clear_entirely_local_costmap.
[bt_navigator-7] [ERROR] [1612853925.971202867] [bt_navigator]: Navigation failed
[bt_navigator-7] [WARN] [1612853925.971368665] [bt_navigator]: [navigate_to_pose] [ActionServer] Aborting handle.
^C[WARNING] [launch]: user interrupted with ctrl-c (SIGINT)
[amcl-2] [INFO] [1612854068.198611099] [rclcpp]: signal_handler(signal_value=2)
[amcl-2] [INFO] [1612854068.204154070] [amcl]: Destroying
[INFO] [recoveries_server-6]: process has finished cleanly [pid 5234]
[INFO] [planner_server-5]: process has finished cleanly [pid 5232]
[INFO] [map_server-1]: process has finished cleanly [pid 5203]
[lifecycle_manager-9] [INFO] [1612854068.198793116] [rclcpp]: signal_handler(signal_value=2)
[... 340 lines cut ...]
    simulate_ahead_time: 2.0
    max_rotational_vel: 1.0
    min_rotational_vel: 0.4
    rotational_acc_lim: 3.2

robot_state_publisher:
  ros__parameters:
    use_sim_time: True

waypoint_follower:
  ros__parameters:
    loop_rate: 20
    stop_on_failure: false
    waypoint_task_executor_plugin: "wait_at_waypoint"   
    wait_at_waypoint:
      plugin: "nav2_waypoint_follower::WaitAtWaypoint"
      enabled: True
      waypoint_pause_duration: 200

```

### Addtional information
[image]
[image]
[image]

SteveMacenski, 2021-03-16:
Did you try now with the new params on `ros2-master` config folder?

sachinkum0009, 2021-03-17:
No, I will try it now.
Thank you for response

closed 2021-04-12 by SteveMacenski

HazemAbidi, 2021-04-29:
Hello, I'm facing similar errors : 
`controller_server-1] [INFO] [1619712384.151567022] [controller_server]: Received a goal, begin computing control effort.
[controller_server-1] [ERROR] [1619712384.655240957] [controller_server]: Extrapolation Error: Lookup would require extrapolation into the future.  Requested time 1619712383.626969 but the latest data is at time 976.434000, when looking up transform from frame [map] to frame [odom]
[controller_server-1] 
[controller_server-1] [ERROR] [1619712384.655287167] [controller_server]: Global Frame: odom Plan Frame size 587: map
[controller_server-1] 
[controller_server-1] [ERROR] [1619712384.655358394] [controller_server]: Could not transform the global plan to the frame of the controller
[controller_server-1] [WARN] [1619712384.655508720] [controller_server_rclcpp_node]: [follow_path] [ActionServer] Aborting handle.
[bt_navigator-4] [ERROR] [1619712384.666088517] [bt_navigator]: Navigation failed
[bt_navigator-4] [WARN] [1619712384.666143659] [bt_navigator]: [navigate_to_pose] [ActionServer] Aborting handle.` 
I'm also using the params in the ros2-master branch.
I'm using ros2 foxy.

sachinkum0009, 2021-05-03:
The issue is related to the transform rate from frame [map] to frame [odom]. Hence the controller server could not transform the global plan to the frame of the controller. @SteveMacenski should we reopen the issue.

HazemAbidi, 2021-05-03:
@sachinkum0009 Have you been able to make it work? If yes How?

sachinkum0009, 2021-05-04:
@HazemAbidi I am also facing same issue. The controller is not able to plan the trajectory.

HazemAbidi, 2021-05-07:
@sachinkum0009 I solved this by changing the global frame from Odom to map and it does the work. However, I wonder how to fix this properly @SteveMacenski?

ajaypaul2008, 2021-09-17:
I think there is a missing TF timeout here https://github.com/rst-tu-dortmund/teb_local_planner/blob/foxy-devel/teb_local_planner/src/teb_local_planner_ros.cpp#L718 adding `tf2::durationFromSec(0.5)` as third argument here solved this issue for me with global_frame as odom for local costmap