rst-tu-dortmund/teb_local_planner #283: [ros2-master] Maintenance and sync with ROS1 branches

doisyg, 2021-03-11:
Hello TEB planners,
After discussion with @SteveMacenski, me and me team will help syncing and putting up to date the `ros2-master` branch (and targeting a release for galactic).
It looks like the last sync with `melodic-devel` has been done one year ago (21/02/2020), furthermore, it doesn't look like `melodic-devel` and `noetic-devel` are in sync after the 17/10/2020.
The fact that the directory structure is not the same between the ROS1 branches and `ros2-master` is not helping with keeping them in sync.

@BriceRenaudeau did the job of tracking the differences between the 3 branches below : 
| **Source** date |`melodic-devel` | `noetic-devel` | `ros2-master` |
| --- |--- |--- | --- |
**08/03/2021**| | | |
| |**Merge  robot stopped before goal reach [#276](https://github.com/rst-tu-dortmund/teb_local_planner/pull/276)**| --- | --- |
| |**Fix free_goal_vel [#250](https://github.com/rst-tu-dortmund/teb_local_planner/pull/250)**|--- | --- |
**04/01/2021**| | | |
| |**Merge use left leg on cross road [#265](https://github.com/rst-tu-dortmund/teb_local_planner/pull/265)**| --- | --- |
| |**Merge divergence detection [#259](https://github.com/rst-tu-dortmund/teb_local_planner/pull/259)**| --- | --- |
| |**Merge fix licence[#254](https://github.com/rst-tu-dortmund/teb_local_planner/pull/254)**| --- | --- |
| |**Merge handle empty polygon[#248](https://github.com/rst-tu-dortmund/teb_local_planner/pull/248)**| --- | --- |
| |**Merge fix doc typo [#246](https://github.com/rst-tu-dortmund/teb_local_planner/pull/246)**|--- | --- |
**08/11/2020**| | | |
| |**Merge clean vertex type [#243](https://github.com/rst-tu-dortmund/teb_local_planner/pull/243)** |--- | --- |
| |**Merge small graph speed up [#244](https://github.com/rst-tu-dortmund/teb_local_planner/pull/244)** | --- | --- |
**25/10/2020**| | | |
| |**Merge updatable robot model [#238](https://github.com/rst-tu-dortmund/teb_local_planner/pull/238)**| --- | --- |
**17/10/2020**| | | |
| | --- | --- |**Merge twist to ackermann [#219](https://github.com/rst-tu-dortmund/teb_local_planner/pull/219)**|
| | --- | --- |**Merge nav2 API update [#233](https://github.com/rst-tu-dortmund/teb_local_planner/pull/233)**|
| | --- |**Merge sync noetic with melodic [#235](https://github.com/rst-tu-dortmund/teb_local_planner/pull/235)**| --- |
| | --- |**Merge limit max feasability check [#231](https://github.com/rst-tu-dortmund/teb_local_planner/pull/231)**| --- |
| | --- |**Merge increase forward max weigh t[#230](https://github.com/rst-tu-dortmund/teb_local_planner/pull/230)**| --- |
| |**Merge fix local minima [#139](https://github.com/rst-tu-dortmund/teb_local_planner/pull/139)**| SYNCED | --- |
| |**Merge fix fatal error [#228](https://github.com/rst-tu-dortmund/teb_local_planner/pull/228)**| SYNCED | --- |
| |**Merge add transform tolerance [#232](https://github.com/rst-tu-dortmund/teb_local_planner/pull/232)**| SYNCED | --- |
**15/07/2020**| | | |
| |**Merge import sys .py [#217](https://github.com/rst-tu-dortmund/teb_local_planner/pull/217)**|  SYNCED | --- |
**02/06/2020**| | | |
| |**Merge new velocity constraint [#210](https://github.com/rst-tu-dortmund/teb_local_planner/pull/210)**| SYNCED | --- |
| |**Merge proportional saturation [#209](https://github.com/rst-tu-dortmund/teb_local_planner/pull/209)**| SYNCED | --- |
| | --- |**Merge Melodic into Noetic [#210](https://github.com/rst-tu-dortmund/teb_local_planner/commit/fd0be413fde3c6caef7f122057f5511be91e531c)**| --- |
**22/06/2020**| | | |
| | --- | --- |**Replace boost usage [#commit](https://github.com/rst-tu-dortmund/teb_local_planner/commit/8bd805e29ef629a003880e22d67d9529da76a87d)**|
| | --- | --- |**Merge fix visualization bug [#212](https://github.com/rst-tu-dortmund/teb_local_planner/pull/212)**|
**29/05/2020**| | | |
| |**Fix RobotFootprintModel visualization bug [#commit](https://github.com/rst-tu-dortmund/teb_local_planner/commit/1693021653ee5bb7a6ec5887e003a4ddd06d16f3)**| SYNCED | --- |
| |**Merge visu circular obstacles [#199](https://github.com/rst-tu-dortmund/teb_local_planner/pull/199)**| SYNCED | --- |
| |**Merge optimize memory alloc [#208](https://github.com/rst-tu-dortmund/teb_local_planner/pull/208)**| SYNCED | --- |
**24/05/2020**| | | |
| |**Add pill shaped obstacle [#commit](https://github.com/rst-tu-dortmund/teb_local_planner/commit/6accc94d07276f17f2d2c1ad769687ac7eb2adef)**| SYNCED | --- |
**12/03/2020**| | | |
| |**Improve 3D H signature [#commit](https://github.com/rst-tu-dortmund/teb_local_planner/commit/b47719309d2f1427653afb7a44d9e293a2fed1c6)**| SYNCED | --- |
| |**Remove normalization [#commit](https://github.com/rst-tu-dortmund/teb_local_planner/commit/8e89a84af519d038bf44246ef2a5fe6306ec1902)**| SYNCED | --- |
**21/02/2020**| | | |
| |--- | --- |**Merge melodic into ROS2 [#commit](https://github.com/rst-tu-dortmund/teb_local_planner/commit/c1934b28a176fc3cf09f2ec5bd250b5a4a10bd11)** |
**27/01/2020**| | | |
| | --- | --- |**Merge eloquent into ROS2 from Steve [#commit](https://github.com/rst-tu-dortmund/teb_local_planner/commit/7ab1a591af536d77b7f8ca82e1722c8a9450b0e1)**  |

I propose the following plan:

1. sync `noetic-devel` with `ros2-master`, check build and basic run, so it will be synced with both ROS1 branches as of 17/10/2020. PR is here: https://github.com/rst-tu-dortmund/teb_local_planner/pull/282
2. then on top, sync `melodic-devel` with `ros2-master`, check build and basic run, so it will be synced with both ROS1 branches as of 08/03/2021
3. Then when everything is synced and running, check that each ROS1 source modification listed above behave as expected in ros2-master

There is also a need to track the PRs targeting ros2-master and to integrate them : https://github.com/rst-tu-dortmund/teb_local_planner/issues/240

SteveMacenski, 2021-03-11:
Sounds good, looks like 14 total things to merge around, 4 from Noetic and the rest from Melodic.

doisyg, 2021-03-14:
Something else to discuss:
https://github.com/rst-tu-dortmund/teb_local_planner/blob/e3acd7d0d7519406cd136a6d8fead0f783025363/teb_local_planner/src/teb_local_planner_ros.cpp#L259-L260
I am not sure `nav2_core::PlannerException` is used appropriatly here (and in the rest of the code), it looks more like a TEB specific exception to me.

SteveMacenski, 2021-03-15:
But this is an exception being thrown back to the `controller server`, not for internal TEB use. It would not be acceptable to throw a plugin-specific exception back to the main server, else that then ties each implementation back to the server breaking encapsulation.

doisyg, 2021-03-16:
> But this is an exception being thrown back to the `controller server`, not for internal TEB use. It would not be acceptable to throw a plugin-specific exception back to the main server, else that then ties each implementation back to the server breaking encapsulation (e.x. you couldn't catch that TEB exception without including TEB headers in the server).

Ok, I get it.

SteveMacenski, 2021-03-16:
@doisyg was https://github.com/rst-tu-dortmund/teb_local_planner/pull/198 synced and should I close it and the ticket with it?

doisyg, 2021-03-16:
> @doisyg was #198 synced and should I close it and the ticket with it?

No, I wanted to keep it separated from the melodic and noetic syncs. It is on my to do list and would like to properly reproduced the problem and test the proposed fix before ( point `3.` on my updated plan above)

doisyg, 2021-03-23:
> @doisyg was #198 synced and should I close it and the ticket with it?

You can close #198 and #203 when  #291 is merged.
I also updated the first post of this ticket for tracking.

SteveMacenski, 2021-04-12:
@doisyg whats the path forward here or are we done?

doisyg, 2021-04-12:
Yes, I guess the next step is to check that each parameter behaves as expected

SteveMacenski, 2021-05-20:
@doisyg how are things going? What's the status of this? Are we done / testing look good? Would this be ready for release to foxy / galactic?

doisyg, 2021-05-21:
I did not tested it as extensively as I hoped yet, but I use it everyday on my robots and it behaves at least as good as in ROS1 with my config. So I would say it is okay to release for galactic (foxy would need some nav2 API change I believe).
I think there is an issue with the visualization of obstacles from the costmap converter that I need to investigate (but it is only visualization, not affecting the behavior of the robot).
Also, I need to check and sync with the 2 recent melodic commits :  https://github.com/rst-tu-dortmund/teb_local_planner/commit/33253193610c7784808def198a9652f2f6f6f1e6 and https://github.com/rst-tu-dortmund/teb_local_planner/commit/073e4c52ccf2ef3679c71a98b03cf8f0bec93e76. Will do early next week at the latest.

SteveMacenski, 2021-05-24:
Awesome! I’ll plan on a release after that’s patched (and the on going costmap converter lifecycle topic)! I very much appreciate your help in standing up TEB in ROS2. I myself am not a user, but I know how important it is for others migrating from ROS1.