rst-tu-dortmund/teb_local_planner #296: /teb_local_planner/src/optimal_planner.cpp :make error

yesterdai123, 2021-04-15:
when I run catkin_make ,it occurs following:
error: cannot convert ‘g2o::HyperGraphElementCreator<teb_local_planner::VertexPose>*’
How can I solve it?

turtleMaster20, 2021-04-15:
I have the same issue. I tried to install g2o but the problem still persists.

If you just want to only use the package without any modification then you can install it as binaries:

1. `sudo apt install ros-[$ROS_VERSION]-teb-local-planner`
2. Remove the package from catkin workspace.

I wanted to use the package with move_base_flex that is why I am trying to install [this](https://github.com/magazino/teb_local_planner/tree/mbf_plugin) version.

turtleMaster20, 2021-04-16:
I installed g2o and all it's dependencies [(here)](https://github.com/RainerKuemmerle/g2o), restarted my system and tried to install the package from source again. This time it worked. **ROS Melodic**.

This issue can now be closed.

sulakshanarya, 2021-09-10:
**Initially there is no directory found.** 

In file included from /usr/local/include/g2o/core/base_binary_edge.h:30,
                 from /home/srarya/catkin_ws/src/teb_local_planner/include/teb_local_planner/g2o_types/base_teb_edges.h:49,
                 from /home/srarya/catkin_ws/src/teb_local_planner/include/teb_local_planner/g2o_types/edge_velocity.h:49,
                 from /home/srarya/catkin_ws/src/teb_local_planner/src/optimal_planner.cpp:42:
/usr/local/include/g2o/core/base_fixed_sized_edge.h:37:12: fatal error: ceres/internal/fixed_array.h: No such file or directory
   37 |   #include <ceres/internal/fixed_array.h>
      |            ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
compilation terminated.

**I commented the following lines in "/usr/local/include/g2o/core/base_fixed_sized_edge.h" & "/usr/local/include/g2o/core/base_variable_sized_edge.h"**

//#ifdef G2O_USE_VENDORED_CERES
#include "g2o/EXTERNAL/ceres/fixed_array.h"
//#else
//  #include <ceres/internal/fixed_array.h>
//#endif

**After changes, running the catkin_make, I get following error:**

../catkin_ws/src/teb_local_planner/src/optimal_planner.cpp: In static member function ‘static void teb_local_planner::TebOptimalPlanner::registerG2OTypes()’:
/home/srarya/catkin_ws/src/teb_local_planner/src/optimal_planner.cpp:140:40: error: cannot convert ‘g2o::HyperGraphElementCreator<teb_local_planner::VertexPose>*’ to ‘const std::shared_ptr<g2o::AbstractHyperGraphElementCreator>&’
  140 |   factory->registerType("VERTEX_POSE", new g2o::HyperGraphElementCreator<VertexPose>);
      |                                        ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
      |                                        |
      |                                        g2o::HyperGraphElementCreator<teb_local_planner::VertexPose>*
In file included from /home/srarya/catkin_ws/src/teb_local_planner/include/teb_local_planner/optimal_planner.h:56,
                 from /home/srarya/catkin_ws/src/teb_local_planner/src/optimal_planner.cpp:39:
/usr/local/include/g2o/core/factory.h:63:78: note:   initializing argument 2 of ‘void g2o::Factory::registerType(const string&, const std::shared_ptr<g2o::AbstractHyperGraphElementCreator>&)’
   63 |                     const std::shared_ptr<AbstractHyperGraphElementCreator>& c);
      |                     ~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~^
/home/srarya/catkin_ws/src/teb_local_planner/src/optimal_planner.cpp:141:44: error: cannot convert ‘g2o::HyperGraphElementCreator<teb_local_planner::VertexTimeDiff>*’ to ‘const std::shared_ptr<g2o::AbstractHyperGraphElementCreator>&’
  141 |   factory->registerType("VERTEX_TIMEDIFF", new g2o::HyperGraphElementCreator<VertexTimeDiff>);
      |                                            ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
      |                                            |
      |                                            g2o::HyperGraphElementCreator<teb_local_planner::VertexTimeDiff>*
In file included from /home/srarya/catkin_ws/src/teb_local_planner/include/teb_local_planner/optimal_planner.h:56,
                 from /home/srarya/catkin_ws/src/teb_local_planner/src/optimal_planner.cpp:39:
/usr/local/include/g2o/core/factory.h:63:78: note:   initializing argument 2 of ‘void g2o::Factory::registerType(const string&, const std::shared_ptr<g2o::AbstractHyperGraphElementCreator>&)’
   63 |                     const std::shared_ptr<AbstractHyperGraphElementCreator>& c);
      |                     ~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~^
/home/srarya/catkin_ws/src/teb_local_planner/src/optimal_planner.cpp:143:46: error: cannot convert ‘g2o::HyperGraphElementCreator<teb_local_planner::EdgeTimeOptimal>*’ to ‘const std::shared_ptr<g2o::AbstractHyperGraphElementCreator>&’
  143 |   factory->registerType("EDGE_TIME_OPTIMAL", new g2o::HyperGraphElementCreator<EdgeTimeOptimal>);
      |                                              ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
      |                                              |
      |                                              g2o::HyperGraphElementCreator<teb_local_planner::EdgeTimeOptimal>*
In file included from /home/srarya/catkin_ws/src/teb_local_planner/include/teb_local_planner/optimal_planner.h:56,
                 from /home/srarya/catkin_ws/src/teb_local_planner/src/optimal_planner.cpp:39:
/usr/local/include/g2o/core/factory.h:63:78: note:   initializing argument 2 of ‘void g2o::Factory::registerType(const string&, const std::shared_ptr<g2o::AbstractHyperGraphElementCreator>&)’
   63 |                     const std::shared_ptr<AbstractHyperGraphElementCreator>& c);
      |                     ~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~^
/home/srarya/catkin_ws/src/teb_local_planner/src/optimal_planner.cpp:144:47: error: cannot convert ‘g2o::HyperGraphElementCreator<teb_local_planner::EdgeShortestPath>*’ to ‘const std::shared_ptr<g2o::AbstractHyperGraphElementCreator>&’
  144 |   factory->registerType("EDGE_SHORTEST_PATH", new g2o::HyperGraphElementCreator<EdgeShortestPath>);
      |                                               ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
      |                                               |
      |                                               g2o::HyperGraphElementCreator<teb_local_planner::EdgeShortestPath>*
[... cut ...]
      |                                               g2o::HyperGraphElementCreator<teb_local_planner::EdgePreferRotDir>*
In file included from /home/srarya/catkin_ws/src/teb_local_planner/include/teb_local_planner/optimal_planner.h:56,
                 from /home/srarya/catkin_ws/src/teb_local_planner/src/optimal_planner.cpp:39:
/usr/local/include/g2o/core/factory.h:63:78: note:   initializing argument 2 of ‘void g2o::Factory::registerType(const string&, const std::shared_ptr<g2o::AbstractHyperGraphElementCreator>&)’
   63 |                     const std::shared_ptr<AbstractHyperGraphElementCreator>& c);