rst-tu-dortmund/teb_local_planner #303: Make fails on Noetic-devel

wittenator, 2021-05-16:
Hi TEB team,

thank you for your exceptional software! 
I am currently trying to move our robotics stack from Melodic to Noetic. But when I build my project with the noetic-devel branch, I get the following errors:

`/home/fe/korjakow/Projects/catkin_ws/src/forks/navigation/local_planner/teb/src/teb_config.cpp: In member function ‘void teb_local_planner::TebConfig::reconfigure(teb_local_planner::TebLocalPlannerReconfigureConfig&)’:
/home/fe/korjakow/Projects/catkin_ws/src/forks/navigation/local_planner/teb/src/teb_config.cpp:196:57: error: ‘class teb_local_planner::TebLocalPlannerReconfigureConfig’ has no member named ‘feasibility_check_lookahead_distance’
  196 |   trajectory.feasibility_check_lookahead_distance = cfg.feasibility_check_lookahead_distance;
      |                                                         ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
/home/fe/korjakow/Projects/catkin_ws/src/forks/navigation/local_planner/teb/src/teb_config.cpp:210:43: error: ‘class teb_local_planner::TebLocalPlannerReconfigureConfig’ has no member named ‘use_proportional_saturation’
  210 |   robot.use_proportional_saturation = cfg.use_proportional_saturation;
      |                                           ^~~~~~~~~~~~~~~~~~~~~~~~~~~
/home/fe/korjakow/Projects/catkin_ws/src/forks/navigation/local_planner/teb/src/teb_config.cpp:228:52: error: ‘class teb_local_planner::TebLocalPlannerReconfigureConfig’ has no member named ‘obstacle_proximity_ratio_max_vel’
  228 |   obstacles.obstacle_proximity_ratio_max_vel = cfg.obstacle_proximity_ratio_max_vel;
      |                                                    ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
/home/fe/korjakow/Projects/catkin_ws/src/forks/navigation/local_planner/teb/src/teb_config.cpp:229:50: error: ‘class teb_local_planner::TebLocalPlannerReconfigureConfig’ has no member named ‘obstacle_proximity_lower_bound’
  229 |   obstacles.obstacle_proximity_lower_bound = cfg.obstacle_proximity_lower_bound;
      |                                                  ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
/home/fe/korjakow/Projects/catkin_ws/src/forks/navigation/local_planner/teb/src/teb_config.cpp:230:50: error: ‘class teb_local_planner::TebLocalPlannerReconfigureConfig’ has no member named ‘obstacle_proximity_upper_bound’
  230 |   obstacles.obstacle_proximity_upper_bound = cfg.obstacle_proximity_upper_bound;
      |                                                  ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
/home/fe/korjakow/Projects/catkin_ws/src/forks/navigation/local_planner/teb/src/teb_config.cpp:253:46: error: ‘class teb_local_planner::TebLocalPlannerReconfigureConfig’ has no member named ‘weight_velocity_obstacle_ratio’
  253 |   optim.weight_velocity_obstacle_ratio = cfg.weight_velocity_obstacle_ratio;
      |                                              ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
/home/fe/korjakow/Projects/catkin_ws/src/forks/navigation/local_planner/teb/src/teb_config.cpp:261:47: error: ‘class teb_local_planner::TebLocalPlannerReconfigureConfig’ has no member named ‘max_number_plans_in_current_class’
  261 |   hcp.max_number_plans_in_current_class = cfg.max_number_plans_in_current_class;
      |                                               ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
/home/fe/korjakow/Projects/catkin_ws/src/forks/navigation/local_planner/teb/src/teb_config.cpp:267:44: error: ‘class teb_local_planner::TebLocalPlannerReconfigureConfig’ has no member named ‘selection_dropping_probability’
  267 |   hcp.selection_dropping_probability = cfg.selection_dropping_probability;
      |                                            ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
make[2]: *** [forks/navigation/local_planner/teb/CMakeFiles/teb_local_planner.dir/build.make:128: forks/navigation/local_planner/teb/CMakeFiles/teb_local_planner.dir/src/teb_config.cpp.o] Error 1
`

Is it possible that the TebLocalPlannerReconfigureConfig is generated incorrectly?

ljburtz, 2021-07-03:
@wittenator did you manage to fix this on Noetic?

wittenator, 2021-07-19:
Hi @ljburtz ,

I ended up using the Melodic version on Noetic which seems to work as far as I see. At least it didn't throw any errors yet.

ljburtz, 2021-07-19:
same here. fingers crossed this isn't the cause of any issues. I'm not knowledgeable enough to attempt fixing the Noetic make

r0gi, 2022-03-03:
I came across the same issue. It turns out I already had the release version of teb_local_planner installed (via apt), so catkin_make used the installed header files, which don't have those parameters (the noetic release is from mid 2020).

```
sudo apt remove ros-noetic-teb-local-planner
```
then rebuild and it should work