rst-tu-dortmund/teb_local_planner #304: How to re-configure footprint completely

ssr-yuki, 2021-05-19:
Hi. I would like to ask you a question related to `footprint` re-configuration.

We are developing a robot that changes its own horizontal lengths during operations. For example, the robot is small under transportation from A to B, it extends and make its body larger on the place B, and goes back to A. Therefore, we desire to update its footprint on B.

[The document](http://wiki.ros.org/teb_local_planner) says that setting `is_footprint_dynamic` to `true` is enough to change `footprint` dynamically.
However, I found that `TebLocalPlannerROS` class has two footprints variable, i.e. `planner->robot_model_` and `footprint_spec_`, and they look independent to each other. 
`is_footprint_dynamic` flag only turns on the latter re-configuration, which synchronizes costmap re-configure.

In my understanding, if I change footprint completely, I have to change both of them with the following procedure:
- set `is_footprint_dynamic` `true` (when I boot a planner)
- use dynamic_reconfigure for local costmap footprint to change `footprint_spec_`
- set new `footprint_model` in the planner namespace on rosparam, then use dynamic_reconfigure for some planner's parameters in order to call `TebLocalPlannerROS::reconfigureCB` that updates `planner->robot_model_`

Is this procedure correct?

IvanVN, 2023-11-10:
Did you manage to do it? I'm trying to do exactly this without success. The costmap footprint is updated and Teb uses it for feasibility, but the footprint model is not updated even though the params in the parameter server are updated and the reconfigure callback triggered.

JiaoxianDu, 2024-08-26:
My solution is to reconfigure `/move_base/global (local) _costmap/robot_radius`, `/move_base/global (local) _costmap/inflation_layer/inflation_radius` and update rosparam `/move_base/TebLocalPlannerROS/footprint_model/radius`, and trigger reconfiguration of teb to call `getRobotFootprintFromParamServer` for updating.
By the way, param `is_footprint_dynamic` seems not to work well, because it tries to get the value of `padded_footprint_`, which won't update until 'footprint_padding' changes value.