rst-tu-dortmund/teb_local_planner #305: Error while reading params file

vimalrajayyappan, 2021-05-19:
Following are the parameters I'm using for TEB Local Planner.
- While using circular footprint model there were no issues.
   the params are:
            plugin: "teb_local_planner::TebLocalPlannerROS"
            footprint_model.type: circular
            footprint_model.radius: 2.3

But when I switch to polygon model I got error as :
what():  failed to initialize rcl: Couldn't parse params file:

The params I'm using are
      plugin: "teb_local_planner::TebLocalPlannerROS"
      footprint_model.type: polygon  
      footprint_model.vertices: [[-0.27, 0.315], [0.27, 0.315], [0.27, -0.315], [-0.27, -0.315]]
      min_obstacle_dist: 0.5
      inflation_dist: 1.0 #0.1
      # costmap_converter_plugin: "costmap_converter::CostmapToPolygonsDBSMCCH"
      costmap_converter_plugin: ""
      costmap_converter_spin_thread: True
      costmap_converter_rate: 15
      enable_homotopy_class_planning: False
      enable_multithreading: True
      optimization_verbose: False
      teb_autoresize: True
      min_samples: 5
      max_samples: 20
      max_global_plan_lookahead_dist: 2.5 #2.0
      visualize_hc_graph: True
      max_vel_x: 1.3 #0.26
      max_vel_theta: 1.0  #1.0
      acc_lim_x: 1.8
      acc_lim_theta: 1.5
      min_turning_radius: 3.0
      transform_tolerance: 0.2
      yaw_goal_tolerance: 0.05 
      xy_goal_tolerance: 0.1

Looking for help. Thanks in advance! :)

vimalrajayyappan, 2021-05-24:
Hi team, Please have a look and kindly provide your valuable advise.

Natthawe, 2022-01-06:
footprint_model:
        type: "polygon"
        vertices: "[[-0.27, 0.315], [0.27, 0.315], [0.27, -0.315], [-0.27, -0.315]]"