rst-tu-dortmund/teb_local_planner #313: Build fails for foxy-devel/ros2-master/dashing-devel with g2o related incompatibilities

saadehmd, 2021-07-08:
**Tried building teb_local_planner in all  branches and it always ends up given these errors in common:**

/home/saad/navigation2_ws/src/navigation2/teb_local_planner/teb_local_planner/src/optimal_planner.cpp: In static member function ‘static void teb_local_planner::TebOptimalPlanner::registerG2OTypes()’:
/home/saad/navigation2_ws/src/navigation2/teb_local_planner/teb_local_planner/src/optimal_planner.cpp:126:40: error: cannot convert ‘g2o::HyperGraphElementCreator<teb_local_planner::VertexPose>*’ to ‘const std::shared_ptr<g2o::AbstractHyperGraphElementCreator>&’
  126 |   factory->registerType("VERTEX_POSE", new g2o::HyperGraphElementCreator<VertexPose>);
      |                                        ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
      |                                        |
      |                                        g2o::HyperGraphElementCreator<teb_local_planner::VertexPose>*
In file included from /home/saad/navigation2_ws/src/navigation2/teb_local_planner/teb_local_planner/include/teb_local_planner/optimal_planner.h:56,
                 from /home/saad/navigation2_ws/src/navigation2/teb_local_planner/teb_local_planner/src/optimal_planner.cpp:42:
/usr/local/include/g2o/core/factory.h:63:78: note:   initializing argument 2 of ‘void g2o::Factory::registerType(const string&, const std::shared_ptr<g2o::AbstractHyperGraphElementCreator>&)’
   63 |                     const std::shared_ptr<AbstractHyperGraphElementCreator>& c);
      |                     ~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~^
/home/saad/navigation2_ws/src/navigation2/teb_local_planner/teb_local_planner/src/optimal_planner.cpp:127:44: error: cannot convert ‘g2o::HyperGraphElementCreator<teb_local_planner::VertexTimeDiff>*’ to ‘const std::shared_ptr<g2o::AbstractHyperGraphElementCreator>&’
  127 |   factory->registerType("VERTEX_TIMEDIFF", new g2o::HyperGraphElementCreator<VertexTimeDiff>);
      |                                            ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
      |                                            |
      |                                            g2o::HyperGraphElementCreator<teb_local_planner::VertexTimeDiff>*
In file included from /home/saad/navigation2_ws/src/navigation2/teb_local_planner/teb_local_planner/include/teb_local_planner/optimal_planner.h:56,
                 from /home/saad/navigation2_ws/src/navigation2/teb_local_planner/teb_local_planner/src/optimal_planner.cpp:42:
/usr/local/include/g2o/core/factory.h:63:78: note:   initializing argument 2 of ‘void g2o::Factory::registerType(const string&, const std::shared_ptr<g2o::AbstractHyperGraphElementCreator>&)’
   63 |                     const std::shared_ptr<AbstractHyperGraphElementCreator>& c);
      |                     ~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~^
/home/saad/navigation2_ws/src/navigation2/teb_local_planner/teb_local_planner/src/optimal_planner.cpp:129:46: error: cannot convert ‘g2o::HyperGraphElementCreator<teb_local_planner::EdgeTimeOptimal>*’ to ‘const std::shared_ptr<g2o::AbstractHyperGraphElementCreator>&’
  129 |   factory->registerType("EDGE_TIME_OPTIMAL", new g2o::HyperGraphElementCreator<EdgeTimeOptimal>);
      |                                              ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
      |                                              |
      |                                              g2o::HyperGraphElementCreator<teb_local_planner::EdgeTimeOptimal>*
In file included from /home/saad/navigation2_ws/src/navigation2/teb_local_planner/teb_local_planner/include/teb_local_planner/optimal_planner.h:56,
                 from /home/saad/navigation2_ws/src/navigation2/teb_local_planner/teb_local_planner/src/optimal_planner.cpp:42:
/usr/local/include/g2o/core/factory.h:63:78: note:   initializing argument 2 of ‘void g2o::Factory::registerType(const string&, const std::shared_ptr<g2o::AbstractHyperGraphElementCreator>&)’
   63 |                     const std::shared_ptr<AbstractHyperGraphElementCreator>& c);
      |                     ~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~^
/home/saad/navigation2_ws/src/navigation2/teb_local_planner/teb_local_planner/src/optimal_planner.cpp:130:47: error: cannot convert ‘g2o::HyperGraphElementCreator<teb_local_planner::EdgeShortestPath>*’ to ‘const std::shared_ptr<g2o::AbstractHyperGraphElementCreator>&’
  130 |   factory->registerType("EDGE_SHORTEST_PATH", new g2o::HyperGraphElementCreator<EdgeShortestPath>);
      |                                               ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
      |                                               |
      |                                               g2o::HyperGraphElementCreator<teb_local_planner::EdgeShortestPath>*
In file included from /home/saad/navigation2_ws/src/navigation2/teb_local_planner/teb_local_planner/include/teb_local_planner/optimal_planner.h:56,
                 from /home/saad/navigation2_ws/src/navigation2/teb_local_planner/teb_local_planner/src/optimal_planner.cpp:42:
/usr/local/include/g2o/core/factory.h:63:78: note:   initializing argument 2 of ‘void g2o::Factory::registerType(const string&, const std::shared_ptr<g2o::AbstractHyperGraphElementCreator>&)’
   63 |                     const std::shared_ptr<AbstractHyperGraphElementCreator>& c);
      |                     ~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~^
/home/saad/navigation2_ws/src/navigation2/teb_local_planner/teb_local_planner/src/optimal_planner.cpp:131:42: error: cannot convert ‘g2o::HyperGraphElementCreator<teb_local_planner::EdgeVelocity>*’ to ‘const std::shared_ptr<g2o::AbstractHyperGraphElementCreator>&’
  131 |   factory->registerType("EDGE_VELOCITY", new g2o::HyperGraphElementCreator<EdgeVelocity>);
      |                                          ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
      |                                          |
      |                                          g2o::HyperGraphElementCreator<teb_local_planner::EdgeVelocity>*
In file included from /home/saad/navigation2_ws/src/navigation2/teb_local_planner/teb_local_planner/include/teb_local_planner/optimal_planner.h:56,
                 from /home/saad/navigation2_ws/src/navigation2/teb_local_planner/teb_local_planner/src/optimal_planner.cpp:42:
/usr/local/include/g2o/core/factory.h:63:78: note:   initializing argument 2 of ‘void g2o::Factory::registerType(const string&, const std::shared_ptr<g2o::AbstractHyperGraphElementCreator>&)’
   63 |                     const std::shared_ptr<AbstractHyperGraphElementCreator>& c);
      |                     ~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~^
/home/saad/navigation2_ws/src/navigation2/teb_local_planner/teb_local_planner/src/optimal_planner.cpp:132:52: error: cannot convert ‘g2o::HyperGraphElementCreator<teb_local_planner::EdgeVelocityHolonomic>*’ to ‘const std::shared_ptr<g2o::AbstractHyperGraphElementCreator>&’
  132 |   factory->registerType("EDGE_VELOCITY_HOLONOMIC", new g2o::HyperGraphElementCreator<EdgeVelocityHolonomic>);
      |                                                    ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
      |                                                    |
      |                                                    g2o::HyperGraphElementCreator<teb_local_planner::EdgeVelocityHolonomic>*
In file included from /home/saad/navigation2_ws/src/navigation2/teb_local_planner/teb_local_planner/include/teb_local_planner/optimal_planner.h:56,
                 from /home/saad/navigation2_ws/src/navigation2/teb_local_planner/teb_local_planner/src/optimal_planner.cpp:42:
/usr/local/include/g2o/core/factory.h:63:78: note:   initializing argument 2 of ‘void g2o::Factory::registerType(const string&, const std::shared_ptr<g2o::AbstractHyperGraphElementCreator>&)’
   63 |                     const std::shared_ptr<AbstractHyperGraphElementCreator>& c);
      |                     ~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~^
/home/saad/navigation2_ws/src/navigation2/teb_local_planner/teb_local_planner/src/optimal_planner.cpp:133:46: error: cannot convert ‘g2o::HyperGraphElementCreator<teb_local_planner::EdgeAcceleration>*’ to ‘const std::shared_ptr<g2o::AbstractHyperGraphElementCreator>&’
  133 |   factory->registerType("EDGE_ACCELERATION", new g2o::HyperGraphElementCreator<EdgeAcceleration>);
[... cut ...]
make[2]: *** [CMakeFiles/teb_local_planner.dir/build.make:102: CMakeFiles/teb_local_planner.dir/src/visualization.cpp.o] Error 1
make[2]: *** [CMakeFiles/teb_local_planner.dir/build.make:141: CMakeFiles/teb_local_planner.dir/src/homotopy_class_planner.cpp.o] Error 1
make[2]: *** [CMakeFiles/teb_local_planner.dir/build.make:76: CMakeFiles/teb_local_planner.dir/src/optimal_planner.cpp.o] Error 1
make[1]: *** [CMakeFiles/Makefile2:78: CMakeFiles/teb_local_planner.dir/all] Error 2
make: *** [Makefile:141: all] Error 2

jamestkpoon, 2021-08-10:
I had the same problem. What fixed it for me was making this fn:

template <typename T>
void register_type(g2o::Factory* factory, const std::string name)
{
  std::unique_ptr<g2o::HyperGraphElementCreator<T>> ptr_(new g2o::HyperGraphElementCreator<T>());
  std::shared_ptr<g2o::HyperGraphElementCreator<T>> shared_(std::move(ptr_));

  factory->registerType(name, shared_);
}

and then e.g. factory->registerType("EDGE_PREFER_ROTDIR", new g2o::HyperGraphElementCreator<EdgePreferRotDir>); becomes register_type<EdgePreferRotDir>(factory, "EDGE_PREFER_ROTDIR");

closed 2021-11-26 by SteveMacenski

spok7, 2023-10-24:
> template < typename T> void register_type(g2o::Factory* factory, const std::string name) { std::unique_ptr<g2o::HyperGraphElementCreator> ptr_(new g2o::HyperGraphElementCreator()); std::shared_ptr<g2o::HyperGraphElementCreator> shared_(std::move(ptr_));
> 
> factory->registerType(name, shared_); }

I'm not a c++ guy, so it took me a few hours before realizing the \<T\> got formatted away because of markdown.
For fellow copy-pasters struggling on Melodic/Noetic, here's the corrected code:
```cpp
template <typename T>
void register_type(g2o::Factory* factory, const std::string name)
{
  std::unique_ptr<g2o::HyperGraphElementCreator<T>> ptr_(new g2o::HyperGraphElementCreator<T>());
  std::shared_ptr<g2o::HyperGraphElementCreator<T>> shared_(std::move(ptr_));

  factory->registerType(name, shared_);
}
```
...or better yet, [just look at the commit below the comment I'm quoting](https://github.com/Tractonomy/teb_local_planner/commit/318372bb7747ee03bac095245a69f405c5dfe018).