rst-tu-dortmund/teb_local_planner #317: The path calculated by TEB will pass through obstacles

Today20180409, 2021-08-08:
When using teb, I found that obstacles have a big impact on its results. Sometimes the path goes straight through obstacles. When passing through the door, the path didn’t go in the middle of the door, but turned to one side. This is frustrating, I tried the costmap_converter plugin, but it still doesn’t work.
[image]

[image]

My parameters are as follows：
TebLocalPlannerROS:
 #Miscellaneous Parameters
 odom_topic: odom                    
 map_frame: map                      
    
 # Trajectory
 teb_autosize: True                                 
 dt_ref: 0.3                                        
 dt_hysteresis: 0.1                                 
 min_samples: 3                                      
 max_samples: 500                                    
 global_plan_overwrite_orientation: True             
 allow_init_with_backwards_motion: False            
 global_plan_viapoint_sep: 1.0                     
 via_points_ordered: true                           
 max_global_plan_lookahead_dist: 3.0               
 global_plan_prune_distance: 1                       
 exact_arc_length: false                            
 force_reinit_new_goal_dist: 1                      
 #force_reinit_new_goal_angular: 0.5 * M_PI         
 feasibility_check_no_poses: 2                            
 publish_feedback: false                             
 #min_resolution_collision_check_angular: M_PI       
 control_look_ahead_poses: 1                        
    
 # Robot     
 max_vel_x: 0.4                                      
 max_vel_x_backwards: 0.10001                          
 max_vel_y: 0.0                                        
 max_vel_theta: 0.5                                  
 acc_lim_x: 0.6                                     
 acc_lim_y: 0                                        
 acc_lim_theta: 1.0                                  
 min_turning_radius: 0.0                                    
 #wheelbase: 1.0                                      
 #cmd_angle_instead_rotvel: false                     
 is_footprint_dynamic: false                        

 footprint_model:                                    
   type: "polygon"                                    
   vertices: [ [0.28, 0.24], [0.28, -0.24], [-0.28, -0.24], [-0.28, 0.24] ]   # for type "polygon"                 

 # GoalTolerance 
 xy_goal_tolerance: 0.2                               
 yaw_goal_tolerance: 0.1                              
 free_goal_vel: false                                 
 complete_global_plan: false                         
                                                      
 # Obstacles
 min_obstacle_dist: 0.01                              
 inflation_dist: 0.3                                 
 dynamic_obstacle_inflation_dist: 0.3                
 include_dynamic_obstacles: true                      
 include_costmap_obstacles: true                     
 costmap_obstacles_behind_robot_dist: 1.0            
 obstacle_poses_affected: 30     
                    
 legacy_obstacle_association: false                                                                      
 obstacle_association_force_inclusion_factor: 10           
 obstacle_association_cutoff_factor: 50                                                              
 
 #costmap_converter_plugin: ""                         
 #costmap_converter_spin_thread: true                  
 #costmap_converter_rate: 5                           

 # Optimization   
 no_inner_iterations: 5                               
 no_outer_iterations: 4                              
 optimization_activate: true                          
 optimization_verbose: false                         
 penalty_epsilon: 0.1                                 
 obstacle_cost_exponent: 4                            
 weight_max_vel_x: 0.8                                
 weight_max_vel_theta: 1                             
 weight_acc_lim_x: 1                                 
 weight_acc_lim_theta: 1                             
 weight_kinematics_nh: 1000                            
                                                      
 weight_kinematics_forward_drive: 1000                
 weight_kinematics_turning_radius: 0                 
 weight_optimaltime: 1 # must be > 0                
 weight_shortest_path: 0                            
 weight_obstacle: 20                                  
 weight_inflation: 0.2                               
 weight_dynamic_obstacle: 1                           
 weight_dynamic_obstacle_inflation: 0.2             
 weight_viapoint: 1                                   
 weight_adapt_factor: 2                              
                                                     
 # Homotopy Class Planner
 enable_homotopy_class_planning: false                               

# Recovery
 
 shrink_horizon_backup: false
 shrink_horizon_min_duration: 10
 oscillation_recovery: true
 oscillation_v_eps: 0.1
 oscillation_omega_eps: 0.1
 oscillation_recovery_min_duration: 10
 oscillation_filter_duration: 10

Can someone give me some suggestions? thanks. @Christoph Rösmann

hxj0316, 2021-11-03:
Hello, I followed the steps to download and compile the teb local path planning algorithm. I would like to ask how to use it for gazebo simulation and how to use it for actual mobile robots?

VRichardJP, 2021-12-07:
I had a similar issue using `include_dynamic_obstacles=true`. It seems that teb behavior changes quite a lot when `include_dynamic_obstacles` is set to `true`. Interestingly the implementation is faster, but more buggy.
Maybe try again without dynamic obstacles?