rst-tu-dortmund/teb_local_planner #319: Costmap_converter_msgsConfig not found

rythm07, 2021-09-09:
Hello,

I'm having some difficulties putting TEB plugin to work with ROS2 foxy. I cloned the repository to my workspace folder and when I try to compile it, it throughs an error saying that it cannot find costmap_converter_msgs. 

Can anyone explain to me how can I resolve this issue?

The screenshot of the error is below:

[image]

Thanks a lot for your help!

closed 2021-09-09 by rythm07

saaangooooo, 2021-10-05:
Hello @rythm07 
I have the same problem.
Could you please tell me how you solved the problem?

chinabbz, 2021-10-13:
I solve this problem by downloading the https://github.com/rst-tu-dortmund/costmap_converter.git and compile with teb