rst-tu-dortmund/teb_local_planner #326: Build fails for foxy-devel/ros2 with g2o related incompatibilities

matteoderose, 2021-11-10:
I'm trying to build teb_local_planner with ROS2 Foxy after having built g2o library. However, I get the following error when I try to build the package:

`Starting >>> teb_local_planner
[Processing: teb_local_planner]                               
--- stderr: teb_local_planner                                 
In file included from /usr/local/include/g2o/core/base_variable_sized_edge.h:111,
                 from /usr/local/include/g2o/core/base_multi_edge.h:30,
                 from /home/matteodr/ros2_ws/src/teb_local_planner/include/teb_local_planner/g2o_types/base_teb_edges.h:51,
                 from /home/matteodr/ros2_ws/src/teb_local_planner/include/teb_local_planner/g2o_types/edge_velocity.h:49,
                 from /home/matteodr/ros2_ws/src/teb_local_planner/include/teb_local_planner/optimal_planner.h:63,
                 from /home/matteodr/ros2_ws/src/teb_local_planner/src/optimal_planner.cpp:42:
/usr/local/include/g2o/core/base_variable_sized_edge.hpp: In member function ‘virtual void g2o::BaseVariableSizedEdge<D, E>::linearizeOplus()’:
/usr/local/include/g2o/core/base_variable_sized_edge.hpp:84:12: error: ‘class ceres::internal::FixedArray<double>’ has no member named ‘fill’
   84 |     add_vi.fill(0.);
      |            ^~~~
/usr/local/include/g2o/core/base_variable_sized_edge.hpp:88:24: error: ‘class ceres::internal::FixedArray<double>’ has no member named ‘data’
   88 |       vi->oplus(add_vi.data());
      |                        ^~~~
/usr/local/include/g2o/core/base_variable_sized_edge.hpp:94:24: error: ‘class ceres::internal::FixedArray<double>’ has no member named ‘data’
   94 |       vi->oplus(add_vi.data());
      |                        ^~~~
In file included from /usr/local/include/g2o/core/base_variable_sized_edge.h:111,
                 from /usr/local/include/g2o/core/base_multi_edge.h:30,
                 from /home/matteodr/ros2_ws/src/teb_local_planner/include/teb_local_planner/g2o_types/base_teb_edges.h:51,
                 from /home/matteodr/ros2_ws/src/teb_local_planner/include/teb_local_planner/g2o_types/edge_velocity.h:49,
                 from /home/matteodr/ros2_ws/src/teb_local_planner/include/teb_local_planner/optimal_planner.h:63,
                 from /home/matteodr/ros2_ws/src/teb_local_planner/src/visualization.cpp:41:
/usr/local/include/g2o/core/base_variable_sized_edge.hpp: In member function ‘virtual void g2o::BaseVariableSizedEdge<D, E>::linearizeOplus()’:
/usr/local/include/g2o/core/base_variable_sized_edge.hpp:84:12: error: ‘class ceres::internal::FixedArray<double>’ has no member named ‘fill’
   84 |     add_vi.fill(0.);
      |            ^~~~
/usr/local/include/g2o/core/base_variable_sized_edge.hpp:88:24: error: ‘class ceres::internal::FixedArray<double>’ has no member named ‘data’
   88 |       vi->oplus(add_vi.data());
      |                        ^~~~
/usr/local/include/g2o/core/base_variable_sized_edge.hpp:94:24: error: ‘class ceres::internal::FixedArray<double>’ has no member named ‘data’
   94 |       vi->oplus(add_vi.data());
      |                        ^~~~
/home/matteodr/ros2_ws/src/teb_local_planner/src/optimal_planner.cpp: In static member function ‘static void teb_local_planner::TebOptimalPlanner::registerG2OTypes()’:
/home/matteodr/ros2_ws/src/teb_local_planner/src/optimal_planner.cpp:126:40: error: cannot convert ‘g2o::HyperGraphElementCreator<teb_local_planner::VertexPose>*’ to ‘const std::shared_ptr<g2o::AbstractHyperGraphElementCreator>&’
  126 |   factory->registerType("VERTEX_POSE", new g2o::HyperGraphElementCreator<VertexPose>);
      |                                        ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
      |                                        |
      |                                        g2o::HyperGraphElementCreator<teb_local_planner::VertexPose>*
In file included from /home/matteodr/ros2_ws/src/teb_local_planner/include/teb_local_planner/optimal_planner.h:56,
                 from /home/matteodr/ros2_ws/src/teb_local_planner/src/optimal_planner.cpp:42:
/usr/local/include/g2o/core/factory.h:63:78: note:   initializing argument 2 of ‘void g2o::Factory::registerType(const string&, const std::shared_ptr<g2o::AbstractHyperGraphElementCreator>&)’
   63 |                     const std::shared_ptr<AbstractHyperGraphElementCreator>& c);
      |                     ~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~^
/home/matteodr/ros2_ws/src/teb_local_planner/src/optimal_planner.cpp:127:44: error: cannot convert ‘g2o::HyperGraphElementCreator<teb_local_planner::VertexTimeDiff>*’ to ‘const std::shared_ptr<g2o::AbstractHyperGraphElementCreator>&’
  127 |   factory->registerType("VERTEX_TIMEDIFF", new g2o::HyperGraphElementCreator<VertexTimeDiff>);
      |                                            ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
      |                                            |
      |                                            g2o::HyperGraphElementCreator<teb_local_planner::VertexTimeDiff>*
In file included from /home/matteodr/ros2_ws/src/teb_local_planner/include/teb_local_planner/optimal_planner.h:56,
                 from /home/matteodr/ros2_ws/src/teb_local_planner/src/optimal_planner.cpp:42:
/usr/local/include/g2o/core/factory.h:63:78: note:   initializing argument 2 of ‘void g2o::Factory::registerType(const string&, const std::shared_ptr<g2o::AbstractHyperGraphElementCreator>&)’
   63 |                     const std::shared_ptr<AbstractHyperGraphElementCreator>& c);
      |                     ~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~^
/home/matteodr/ros2_ws/src/teb_local_planner/src/optimal_planner.cpp:129:46: error: cannot convert ‘g2o::HyperGraphElementCreator<teb_local_planner::EdgeTimeOptimal>*’ to ‘const std::shared_ptr<g2o::AbstractHyperGraphElementCreator>&’
  129 |   factory->registerType("EDGE_TIME_OPTIMAL", new g2o::HyperGraphElementCreator<EdgeTimeOptimal>);
      |                                              ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
      |                                              |
      |                                              g2o::HyperGraphElementCreator<teb_local_planner::EdgeTimeOptimal>*
In file included from /home/matteodr/ros2_ws/src/teb_local_planner/include/teb_local_planner/optimal_planner.h:56,
                 from /home/matteodr/ros2_ws/src/teb_local_planner/src/optimal_planner.cpp:42:
/usr/local/include/g2o/core/factory.h:63:78: note:   initializing argument 2 of ‘void g2o::Factory::registerType(const string&, const std::shared_ptr<g2o::AbstractHyperGraphElementCreator>&)’
   63 |                     const std::shared_ptr<AbstractHyperGraphElementCreator>& c);
      |                     ~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~^
/home/matteodr/ros2_ws/src/teb_local_planner/src/optimal_planner.cpp:130:47: error: cannot convert ‘g2o::HyperGraphElementCreator<teb_local_planner::EdgeShortestPath>*’ to ‘const std::shared_ptr<g2o::AbstractHyperGraphElementCreator>&’
  130 |   factory->registerType("EDGE_SHORTEST_PATH", new g2o::HyperGraphElementCreator<EdgeShortestPath>);
[... 751 lines cut ...]
/usr/local/include/g2o/core/base_fixed_sized_edge.hpp:179:30: error: ‘FixedArray’ {aka ‘class ceres::internal::FixedArray<double, 3>’} has no member named ‘data’
  179 |     vertex->oplus(add_vertex.data());
      |                   ~~~~~~~~~~~^~~~
/usr/local/include/g2o/core/base_fixed_sized_edge.hpp: In instantiation of ‘void g2o::BaseFixedSizedEdge<D, E, VertexTypes>::linearizeOplusN() [with int N = 1; int D = 2; E = double; VertexTypes = {teb_local_planner::VertexPose, teb_local_planner::VertexPose}]’:
/usr/local/include/g2o/core/base_fixed_sized_edge.hpp:192:43:   required from ‘void g2o::BaseFixedSizedEdge<D, E, VertexTypes>::linearizeOplusNs(std::index_sequence<Ints ...>) [with long unsigned int ...Ints = {0, 1}; int D = 2; E = double; VertexTypes = {teb_local_planner::VertexPose, teb_local_planner::VertexPose}; std::index_sequence<Ints ...> = std::integer_sequence<long unsigned int, 0, 1>]’
/usr/local/include/g2o/core/base_fixed_sized_edge.hpp:200:3:   required from ‘void g2o::BaseFixedSizedEdge<D, E, VertexTypes>::linearizeOplus() [with int D = 2; E = double; VertexTypes = {teb_local_planner::VertexPose, teb_local_planner::VertexPose}]’
/usr/local/include/g2o/core/base_fixed_sized_edge.hpp:197:6:   required from here
[... cut ...]
  179 |     vertex->oplus(add_vertex.data());
      |                   ~~~~~~~~~~~^~~~
make[2]: *** [CMakeFiles/teb_local_planner.dir/build.make:154: CMakeFiles/teb_local_planner.dir/src/teb_local_planner_ros.cpp.o] Error 1
make[1]: *** [CMakeFiles/Makefile2:78: CMakeFiles/teb_local_planner.dir/all] Error 2
make: *** [Makefile:141: all] Error 2
---
Failed   <<< teb_local_planner [35.9s, exited with code 2]

Summary: 12 packages finished [43.2s]
  1 package failed: teb_local_planner
  1 package had stderr output: teb_local_planner`

guilyx, 2021-12-03:
Hey, for me installing g2o using `apt install ros-foxy-libg2o` made the compilation work fine.
If you're building from source and absolutely require that, can't help.

fvillml, 2022-03-25:
Any update on this? I would like to compile it from source too for Ubuntu 20 and I'm having the same problem. I compiled and installed g2o from source too because I don't want to install some packages with the ros apt package.

BIRL-xu, 2022-08-08:
> Any update on this? I would like to compile it from source too for Ubuntu 20 and I'm having the same problem. I compiled and installed g2o from source too because I don't want to install some packages with the ros apt package.

 You could try the commit **565d1000** of g2o master branch, and commit **93511bfd** of Ceres master branch. I built these on Ubuntu 20.04 and Galactic. Hope that help you.

gongbingyu, 2024-05-23:
> You could try the commit **565d1000** of g2o master branch, and commit **93511bfd** of Ceres master branch. I built these on Ubuntu 20.04 and Galactic. Hope that help you.

I'm not find the commit 565d1000 of g2o master branch