rst-tu-dortmund/teb_local_planner #351: `void tf2::fromMsg<geometry_msgs::msg::Quaternion_<std::allocator<void> >, tf2::Quaternion>(geometry_msgs::msg::Quaternion_<std::allocator<void> > const&, tf2::Quaternion&)`  nav2_bringup/composed_bringup symbol lookup error

czg-sci-42ver, 2022-01-20:
I used `ros2 rolling` and cloned `nav2` and `teb_local_planner` into the same repository
I also installed the dependencys

- here is my directory tree
```
08:36:41 czg@czg-Lenovo-Legion-R7000-2020 nav2_ws ±|master ✗|→ tree -L 1 src/
src/
├── build
├── costmap_converter
├── hls_lfcd_lds_driver
├── install
├── log
├── navigation2
├── navigation2_tutorials
├── slam_toolbox
├── teb_local_planner
└── turtlebot3

08:48:50 czg@czg-Lenovo-Legion-R7000-2020 nav2_ws ±|master ✗|→ tree -L 2 src/teb_local_planner/
src/teb_local_planner/
├── README.md
├── teb_local_planner
│   ├── cfg
│   ├── CHANGELOG.rst
│   ├── CMakeLists.txt
│   ├── cmake_modules
│   ├── include
│   ├── launch
│   ├── LICENSE
│   ├── package.xml
│   ├── params
│   ├── scripts
│   ├── src
│   ├── teb_local_planner_plugin.xml
│   └── test
└── teb_msgs
    ├── CMakeLists.txt
    ├── msg
    └── package.xml
```
- I think main problem is

[composed_bringup-9] [INFO] [1642682600.234018235] [bt_navigator]: Begin navigating from current location to (0.38, -0.60)
[composed_bringup-9] [INFO] [1642682600.255151534] [controller_server]: Received a goal, begin computing control effort.
[composed_bringup-9] [WARN] [1642682600.255236810] [controller_server]: No goal checker was specified in parameter 'current_goal_checker'. Server will use only plugin loaded goal_checker . This warning will appear once.
[composed_bringup-9] /home/czg/nav2_ws/install/nav2_bringup/lib/nav2_bringup/composed_bringup: **symbol lookup error: /home/czg/nav2_ws/install/teb_local_planner/lib/libteb_local_planner.so: undefined symbol: _ZN3tf27fromMsgIN13geometry_msgs3msg11Quaternion_ISaIvEEENS_10QuaternionEEEvRKT_RT0_**
[ERROR] [composed_bringup-9]: process has died [pid 48842, exit code 127, cmd '/home/czg/nav2_ws/install/nav2_bringup/lib/nav2_bringup/composed_bringup --ros-args --params-file /tmp/tmp9b7q7zvt --params-file /tmp/launch_params_dyqx3188 -r /tf:=tf -r /tf_static:=tf_static'].

- the function can‘t be found.
```
08:52:22 czg@czg-Lenovo-Legion-R7000-2020 ~ → c++filt _ZN3tf27fromMsgIN13geometry_msgs3msg11Quaternion_ISaIvEEENS_10QuaternionEEEvRKT_RT0_
void tf2::fromMsg<geometry_msgs::msg::Quaternion_<std::allocator<void> >, tf2::Quaternion>(geometry_msgs::msg::Quaternion_<std::allocator<void> > const&, tf2::Quaternion&)
```

- my main package branches are

08:55:55 czg@czg-Lenovo-Legion-R7000-2020 navigation2 ±|**main** ✗|→ cd ../teb_local_planner/
08:57:06 czg@czg-Lenovo-Legion-R7000-2020 teb_local_planner ±|**ros2-master** ✗|→ 

- the tf error is also weird
```
 [composed_bringup-9] [INFO] [1642682593.611071527] [local_costmap.local_costmap]: Timed out waiting for transform from base_link to odom to become available, tf error: Could not find a connection between 'odom' and 'base_link' because they are not part of the same tree.Tf has two or more unconnected trees.

08:44:58 czg@czg-Lenovo-Legion-R7000-2020 ~ → ros2 run tf2_ros tf2_echo odom base_link
[INFO] [1642682705.299311668] [tf2_echo]: Waiting for transform odom ->  base_link: Invalid frame ID "odom" passed to canTransform argument target_frame - frame does not exist
At time 110.223000000
- Translation: [-2.000, -0.500, 0.019]
- Rotation: in Quaternion [0.000, 0.001, 0.003, 1.000]
At time 111.209000000
- Translation: [-2.000, -0.500, 0.019]
- Rotation: in Quaternion [0.000, 0.001, 0.003, 1.000]
^C[INFO] [1642682708.120119739] [rclcpp]: signal_handler(signum=2)
```

czg-sci-42ver, 2022-01-20:
- my launch
```
ros2 launch teb_local_planner teb_tb3_simulation_launch.py 
```

```python
# Copyright (c) 2018 Intel Corporation
#
# Licensed under the Apache License, Version 2.0 (the "License");
# you may not use this file except in compliance with the License.
# You may obtain a copy of the License at
#
#     http://www.apache.org/licenses/LICENSE-2.0
#
# Unless required by applicable law or agreed to in writing, software
# distributed under the License is distributed on an "AS IS" BASIS,
# WITHOUT WARRANTIES OR CONDITIONS OF ANY KIND, either express or implied.
# See the License for the specific language governing permissions and
# limitations under the License.

# This is all-in-one launch script intended for use by nav2 developers.

import os

from ament_index_python.packages import get_package_share_directory

from launch import LaunchDescription
from launch.actions import IncludeLaunchDescription
from launch.actions import DeclareLaunchArgument

from launch.launch_description_sources import PythonLaunchDescriptionSource
from launch.substitutions import LaunchConfiguration
from nav2_common.launch import HasNodeParams, RewrittenYaml

def generate_launch_description():
    # Get the launch directory
    teb_launch_dir = os.path.join(
        get_package_share_directory('teb_local_planner'), 'launch')
    teb_param_dir = os.path.join(
        get_package_share_directory('teb_local_planner'), 'params')

    nav2_bringup_launch_dir = os.path.join(
        get_package_share_directory('nav2_bringup'), 'launch')
    nav2_bringup_rviz_dir = os.path.join(
        get_package_share_directory('nav2_bringup'), 'rviz')
    nav2_bringup_dir =  get_package_share_directory('nav2_bringup')

    lifecycle_nodes = ['map_saver']

    # params_file = os.path.join(teb_param_dir, 'teb_params.yaml')
    # rviz_config_file = os.path.join(teb_rviz_dir, 'rviz_test_optim.rviz')

    # params_file = os.path.join(nav2_bringup_dir, 'params', 'nav2_params.yaml')
    params_file = os.path.join(teb_launch_dir, 'teb_params_add.yaml')
    rviz_config_file = os.path.join(nav2_bringup_rviz_dir,  'nav2_default_view.rviz')

    bringup_cmd = IncludeLaunchDescription(
        PythonLaunchDescriptionSource(os.path.join(nav2_bringup_launch_dir, 'tb3_simulation_launch.py')),
        launch_arguments={
            'params_file': params_file,
            'rviz_config_file': rviz_config_file,
            'slam': 'True'}.items())

    ld = LaunchDescription()
    ld.add_action(bringup_cmd)
    print(ld)

    return ld

```
[... 121 lines cut ...]

      roadmap_graph_no_samples: 15
      roadmap_graph_area_width: 5.0
      roadmap_graph_area_length_scale: 1.0
      h_signature_prescaler: 0.5
      h_signature_threshold: 0.1
      obstacle_heading_threshold: 0.45
      switching_blocking_period: 0.0
      viapoints_all_candidates: True
      delete_detours_backwards: True
      max_ratio_detours_duration_best_duration: 3.0
      visualize_hc_graph: False
      visualize_with_time_as_z_axis_scale: 0.0

      # Recovery
      
      shrink_horizon_backup: True
      shrink_horizon_min_duration: 10.0
      oscillation_recovery: True
      oscillation_v_eps: 0.1
      oscillation_omega_eps: 0.1
      oscillation_recovery_min_duration: 10.0
      oscillation_filter_duration: 10.0

```

czg-sci-42ver, 2022-01-21:
I solved my problem after making the following changes: 
```diff
diff --git a/include/teb_local_planner/pose_se2.h b/include/teb_local_planner/pose_se2.h
index addd0dd..4ec990e 100755
--- a/include/teb_local_planner/pose_se2.h
+++ b/include/teb_local_planner/pose_se2.h
@@ -46,9 +46,10 @@
 #include <geometry_msgs/msg/pose.hpp>
 #include <geometry_msgs/msg/pose2_d.hpp>
 
+#include <tf2_geometry_msgs/tf2_geometry_msgs.h>
+
 #include <tf2/convert.h>
 #include <tf2/utils.h>
-#include <tf2_geometry_msgs/tf2_geometry_msgs.h>
 
 namespace teb_local_planner
 {
```

guilyx, 2022-01-31:
https://github.com/ros2/geometry2/pull/485 is related and potentially fixes it without your changes, but it needs to be merged 😵‍💫

SteveMacenski, 2022-08-24:
This has been merged!

closed 2022-08-24 by SteveMacenski