rst-tu-dortmund/teb_local_planner #352: Unable to figure out  teb_local_planner

DistroIII, 2022-01-23:
Does teb_local_Planner need costmap_converter package to work properly? I tried following the Tutorials to get teb_local_plalnner to work for turtlebot3 but it didnt work, teb_local_planner couldnt find feasible paths to any of the goals I set.Any help would be greatly appreciated. Here is move_base.launch file I editted in order to replace the DWA_local_planner with the teb_local_planner:

```
<launch>
  <!-- Arguments -->
  <arg name="model" default="$(env TURTLEBOT3_MODEL)" doc="model type [burger, waffle, waffle_pi]"/>
  <arg name="cmd_vel_topic" default="/cmd_vel" />
  <arg name="odom_topic" default="odom" />
  <arg name="move_forward_only" default="false"/>

  <!-- move_base -->
  <node pkg="move_base" type="move_base" respawn="false" name="move_base" output="screen">
    <param name="base_local_planner" value="dwa_local_planner/DWAPlannerROS" />
    <rosparam file="$(find turtlebot3_navigation)/param/costmap_common_params_$(arg model).yaml" command="load" ns="global_costmap" />
    <rosparam file="$(find turtlebot3_navigation)/param/costmap_common_params_$(arg model).yaml" command="load" ns="local_costmap" />
    <rosparam file="$(find turtlebot3_navigation)/param/local_costmap_params.yaml" command="load" />
    <rosparam file="$(find turtlebot3_navigation)/param/global_costmap_params.yaml" command="load" />
    <rosparam file="$(find turtlebot3_navigation)/param/move_base_params.yaml" command="load" />
    <rosparam file="$(find turtlebot3_navigation)/param/dwa_local_planner_params_$(arg model).yaml" command="load" />
    <remap from="cmd_vel" to="$(arg cmd_vel_topic)"/>
    <remap from="odom" to="$(arg odom_topic)"/>
    <param name="base_local_planner" value="teb_local_planner/TebLocalPlannerROS" />
    <param name="controller_frequency" value="10.0" />
  </node>
</launch>
```

Someone please tell me if I'm doing something wrong.

DistroIII, 2022-01-27:
I figured it out, never mind although I have naother question which I will shortly ask

r0gi, 2022-03-02:
It doesn't need the costmap_converter to work properly. If teb is configured to use your local costmap (default) then it will treat every occupied pixel in that costmap as a unique obstacle. The costmap_converter instead converts groups of occupied pixels into points or lines, then feeds that into the teb instead of the raw costmap, which can reduce CPU load.

You should explain how you figured out your issue for others who may have the same problem in the future.