rst-tu-dortmund/teb_local_planner #359: Holonomic robot: oscillation when strafing

corot, 2022-03-11:
Looks like when we strafe we don't update the beginning of the trajectory, and so the look-ahead poses gets behind of the robot:

https://user-images.githubusercontent.com/322610/157814078-4f7e7ac2-846e-4c98-8f43-0236b354ad14.mp4

When moving fw/bw the beginning of the trajectory always starts on current robot pose, so there's no problem:

Uploading teb_linear_move_ok.mp4…

corot, 2022-03-15:
After some research, I narrowed down the problem, but still don't have a solution. For some unknown reason, the `dt` of the TEB poses don't get updated when strafing, I guess because the optimization only considers velocity along x axis. I printed the total time for the trajectory and it remains constant!
You can see on the video how the poses height (as a proxy of estimated reaching time) remains static when strafing, but not when moving straight or even when we have a small `vx` component:

https://user-images.githubusercontent.com/322610/158321796-1cfd072c-4d2d-4056-b130-5e57ed6fb273.mp4

Reducing `dt_hysteresis` avoids the issue because we keep dropping the traversed poses even without updating the the `dt`.

corot, 2022-03-24:
Still not fixed, but I narrowed down the problem a bit more. Looks like the [EdgeAccelerationHolonomicStart](https://github.com/rst-tu-dortmund/teb_local_planner/blob/38be03dbd59f81667f12589be1851de08faabc8d/include/teb_local_planner/g2o_types/edge_acceleration.h#L566), as I can avoid the problem by setting _error[1] to 0, or weight_acc_lim_y to 0.

*DISCLAIMER: no idea why this happens*

corot, 2022-10-21:
@doisyg, remember, I invite u to yakiniku!

closed 2023-06-21 by corot

Fuzhiyuan, 2025-07-01:
@corot hi, how do you make the robot move like this? I have tried to modify some Teb`s parameters, such as max_vel_y, weight_kinematics_nh and acc_lim_y, but it still can not strafe like your robot did and its gesture is like a slash even though target is beside it, do you have any ideas?

commit 3b91dcc3 (2023-06-21): Fix #359: Remove penalty_epsilon from max_vel_x error calculation