rst-tu-dortmund/teb_local_planner #360: [Error]After I drag the obstacles, the Rviz crashed:[rviz-3] process has died

2713874948, 2022-03-25:
Sorry for my poor English.
I successfully compiled the TEB package. Then I started to run the package according to the command `~/teb_ws$ roslaunch teb_local_planner test_optim_node`, and the RVIZ window popped up. 
`... logging to /home/wtc/.ros/log/0efb98a2-ac0c-11ec-9c1e-60e32bf2c33f/roslaunch-wtc-Lenovo-Legion-Y9000P2021H-4079.log
Checking log directory for disk usage. This may take a while.
Press Ctrl-C to interrupt
Done checking log file disk usage. Usage is <1GB.

started roslaunch server http://wtc-Lenovo-Legion-Y9000P2021H:37863/

SUMMARY
========

PARAMETERS
 * /rosdistro: melodic
 * /rosversion: 1.14.12

NODES
  /
    rviz (rviz/rviz)
    test_optim_node (teb_local_planner/test_optim_node)

auto-starting new master
process[master]: started with pid [4089]
ROS_MASTER_URI=http://localhost:11311

setting /run_id to 0efb98a2-ac0c-11ec-9c1e-60e32bf2c33f
process[rosout-1]: started with pid [4100]
started core service [/rosout]
process[test_optim_node-2]: started with pid [4104]
process[rviz-3]: started with pid [4108]
[ INFO] [1648192833.809447005]: max_vel_trans is not set, setting it equal to max_vel_x: 0.4
[ INFO] [1648192833.830407361]: No robot footprint model specified for trajectory optimization. Using point-shaped model.`

If I do nothing, the program will not report an error. But once I drag the obstacle in RVIZ, the program will report an error like this:
`[rviz-3] process has died [pid 23098, exit code -11, cmd /opt/ros/melodic/lib/rviz/rviz -d /home/wtc/teb_ws/src/teb_local_planner/cfg/rviz_test_optim.rviz __name:=rviz __log:=/home/wtc/.ros/log/780304c6-ac0b-11ec-9c1e-60e32bf2c33f/rviz-3.log].
log file: /home/wtc/.ros/log/780304c6-ac0b-11ec-9c1e-60e32bf2c33f/rviz-3*.log
`
I looked up this problem on google and didn't find a solution to this problem. I'm new to ros, thank you very much for checking out my question.

2713874948, 2022-03-26:
I found that if the parallel planning is deactivated with the following command
`rosparam set /test_optim_node/enable_homotopy_class_planning False`, 
the program will run very stable.
But once the parallel planning is enabled with the following command
`rosparam set /test_optim_node/enable_homotopy_class_planning True`, 
the program will easily crash when I drag the obstacle.

posshe, 2022-04-13:
@2713874948