rst-tu-dortmund/teb_local_planner #365: saturateVelocity issue in ros2-master

stu00608, 2022-05-03:
Hello, I'm using ROS2 Galactic and running TEB controller in `ros2-master` branch ([6d782b](https://github.com/rst-tu-dortmund/teb_local_planner/commit/6d782bf1deb9c86a6ececeec066601e924006318))

The robot I'm using is a holonomic robot, I found the linear y velocity is over the limit a lot.
`max_vel_y` is 0.25.
[image]
 
So I checked the foxy branch. Apparantly, the issue is because the ratio calculation.
[image]

But it seems been solved in foxy branch. 
[image]

Then I'm not sure, should I keep develop in `ros2-master` branch? In this case, I can create a PR and fix that piece.
Also, will you guys release a galactic branch?

Thank you!

stu00608, 2022-05-05:
I think the reason to calculate the ratio but not simply limit the velocity at maximum value is because the flag `use_proportional_saturation` , but I don't see this parameter in rqt_reconfigure, so is it deprecated?

corot, 2022-05-31:
[This PR](https://github.com/rst-tu-dortmund/teb_local_planner/pull/324/files) solves the problem, right?

If so, I can open the same for all non-fixed branches

stu00608, 2022-06-01:
@corot Yes, that solves the problem, not sure how many branches has this issue but at least in `ros2-master`.

corot, 2022-06-01:
thanks PaRed to noetic and ROS2
- https://github.com/rst-tu-dortmund/teb_local_planner/pull/367
- https://github.com/rst-tu-dortmund/teb_local_planner/pull/368

closed 2022-06-14 by stu00608