rst-tu-dortmund/teb_local_planner #369: robot "oscillation"error occured when reached the goal, but robot doesn't seem to oscillate

jaykorea, 2022-06-10:
Hello, I'm using ros navigtaion stack consist of 'navfn global planner' & 'teb local planner'

Robot driving cource and trajectory looks fine, but teb local planner does not trigger "goal Reached" message for me.

As I set the goal point, robot reach to the goal very smoothly, but local planner message says that " Aborting because the robot appears to be oscillating over and over. Even after executing all recovery behaviors "

I 've check /cmd_vel topic, and I could not find any data that cause robot oscillaton.

Teb local planner seems not getting robot's location that it is arrived. Is it a bug or can fix by param setting?

below is my rviz video and teb local planner param settings

video : https://youtu.be/e4PoNNMUsG0

```javascript
TebLocalPlannerROS:

 odom_topic: odom
 map_frame: map # map
    
 # Trajectory
  
 teb_autosize: True
 dt_ref: 0.3
 dt_hysteresis: 0.11
 global_plan_overwrite_orientation: True # defualt : true
 max_global_plan_lookahead_dist: 2.0 #default 2.0
 feasibility_check_no_poses: 5
 global_plan_viapoint_sep: -0.1
 publish_feedback: True 
 allow_init_with_backwards_motion: false  

 # Robot
         
 max_vel_x: 0.3 # default 0.25
 max_vel_x_backwards: 0.15
 max_vel_theta: 0.35 # default 0.45
 acc_lim_x: 0.4 # default 0.5
 acc_lim_theta: 0.3 # default : 0.4
 min_turning_radius: 0
 footprint_model: # types: "point", "circular", "two_circles", "line", "polygon"
  type: "polygon" #"line"
  vertices: [[-0.35, -0.35], [-0.35, 0.35], [0.35, 0.35], [0.35, -0.35]]
   #radius: 0.2 # for type "circular"
   #line_start: [-0.2, 0.0] # for type "line"
   #line_end: [0.2, 0.0] # for type "line"
   #front_offset: 0.2 # for type "two_circles"
   #front_radius: 0.2 # for type "two_circles"
   #rear_offset: 0.2 # for type "two_circles"
   #rear_radius: 0.2 # for type "two_circles"
   #vertices: [ [0.25, -0.05], [0.18, -0.05], [0.18, -0.18], [-0.19, -0.18], [-0.25, 0], [-0.19, 0.18], [0.18, 0.18], [0.18, 0.05], [0.25, 0.05] ] # for type "polygon"

 # GoalTolerance
    
 xy_goal_tolerance: 0.9
 yaw_goal_tolerance: 0.3
 free_goal_vel: False
    
 # Obstacles
    
 min_obstacle_dist: 0.05 # This value must also include our robot radius, since footprint_model is set to "point" or "line".
 include_costmap_obstacles: True
 costmap_obstacles_behind_robot_dist: 1.0 #default : 1.0
 obstacle_poses_affected: 30
 costmap_converter_plugin: ""
 costmap_converter_spin_thread: True
 costmap_converter_rate: 5

 # Optimization
    
 no_inner_iterations: 5
 no_outer_iterations: 4
 optimization_activate: True
 optimization_verbose: False
 penalty_epsilon: 0.1
 weight_max_vel_x: 2
 weight_max_vel_theta: 1
 weight_acc_lim_x: 1
 weight_acc_lim_theta: 1
 weight_kinematics_nh: 1000
 weight_kinematics_forward_drive: 1
 weight_kinematics_turning_radius: 1
 weight_optimaltime: 1
 weight_obstacle: 50
#  weight_dynamic_obstacle: 10 # not in use yet
#  alternative_time_cost: False # not in use yet

 # Homotopy Class Planner

 enable_homotopy_class_planning: False # I turned it of because of  #[ WARN] [1654158502.701863979]: Control loop missed its desired rate of 10.0000Hz... the loop actually took 0.1477 seconds
 enable_multithreading: True
 simple_exploration: False
 max_number_classes: 4
 roadmap_graph_no_samples: 15
 roadmap_graph_area_width: 5
 h_signature_prescaler: 0.5
 h_signature_threshold: 0.1
 obstacle_keypoint_offset: 0.1
 obstacle_heading_threshold: 0.45
 visualize_hc_graph: False

```

jcmonteiro, 2022-06-10:
I believe you are missing the goal checker plugin. In the parameter section of your controller server, you should have something like (assuming ROS2).

```yaml
controller_server_node_name:
  ros__parameters:
    general_goal_checker:
      plugin: "nav2_controller::SimpleGoalChecker"
```

Should be similar in ROS1.

jcmonteiro, 2022-06-10:
> Thanks for ur super fast reply.

No worries =)

> I'm using ros1 and there is nothing to access to goal checker plugin. Am I missing something?

Hard to say. I have transitioned to ROS2 for a year now.

> Is there any correlation between this problem with controller_frequency??

I don't believe so. Especially since you already had a "reasonable" update frequency.

From your video it appears that TEB has actually reached the goal and stopped, so you could try to debug this from "bottom" to "top", meaning from the planner to the controller server, to identify in which step the information is being lost.

jaykorea, 2022-06-10:
> No worries =)
>
>
> Hard to say. I have transitioned to ROS2 for a year now.
>
>
> I don't believe so. Especially since you already had a "reasonable" update frequency.
>
> From your video it appears that TEB [...]

Thank for your replay again :)

It would be the best way to debug it from bottom to the top, I don't have much time to do it..

Can you tell me which part should I check first?(goal trigger part?)

I have already checked odom data that translating right coordination position

I dont't have any idea why this error occurs ..(but solved with freequency param)

Thank you

anujpatel1608, 2023-04-17:
Hi @jaykorea ,By frequency param you mean the controller frequency or costmap frequency?