rst-tu-dortmund/teb_local_planner #370: How to make robot turn to correct direction firstly?

oym1994, 2022-06-21:
Hi,
Thanks for this great job. I am testing it with my real robot. I have found that, if the goal is behind the robot, it will move backward and turn. I want the robot to change its direction first(rotation in place, the robot minimum turning radius is set to be 0) and then move, finally then change its direction when reached or near the position.How can I change the parameters to achieve that? Or maybe the algorithm dose not support this kind of planning?
Thanks for your attention and wait for your kind response!

oym1994, 2022-06-23:
Thank you for your kind response. I have yet tried [rotation shim](https://navigation.ros.org/configuration/packages/configuring-rotation-shim-controller.html), it really can solve my problem!

jcmonteiro, 2022-06-23:
Glad it helped. I suggest closing the issue if it is indeed resolved.

oym1994, 2022-06-24:
> Glad it helped. I suggest closing the issue if it is indeed resolved.

OK!