rst-tu-dortmund/teb_local_planner #373: [ros2] General ros2 maintenance and Humble support

doisyg, 2022-07-11:
Hello TEB users,
I did not come here for a long time, but it seems that the `noetic` branch have seen lots of activity (@corot , @kenomo , @siferati ).
I believe there is a sync that needs to be done with the `ros2-master` branch, and also the `ros2-master` branch needs changes to work with `humble` (@chenjunn) , @tonynajjar). 
I have already a `humble` version that works with the last nav2 (https://github.com/rst-tu-dortmund/teb_local_planner/pull/371) plus some changes that I never pushed for dynamic footprint support (@guilyx). I am willing to contribute an push these changes in the following days.
I propose that all parties interested discuss here how to get organized. Who has the maintenance rights nowadays ? I believe a first step should be to create a `galactic` branch from a copy of the current `ros2-master`. Then maybe a `humble` and `rolling` branch , or we continue to use `ros2-master` ?

doisyg, 2022-07-18:
First PR for both `humble` support and sharing changes form our fork: https://github.com/rst-tu-dortmund/teb_local_planner/pull/374
Once merged (maybe in a `humble` branch ?), the next step is probably to review and decide to sync if needed the different `noetic` commits

doisyg, 2022-07-18:
Updated PR (never merged) for costmap_converter: https://github.com/rst-tu-dortmund/costmap_converter/pull/27
Compiler warnings, spin_some memory leak workaround and tf2_geometry header.

davidteixeira, 2022-07-29:
Hi @doisyg . I checked out your branch (great work btw, thanks) and tested it on a ROS2 docker container -- everything seems to be working as expected.

How can we move forward to make this "official"? Considering TEB is mentioned as one of the controller plugins for Nav2, I think having an up to date release would be important.

doisyg, 2022-07-29:
Hi @davidteixeira, I don't really know, we need somebody with admin rights. Last time (galactic) @SteveMacenski gave us a hand for merging but we did not make an official release.

automech-rb, 2022-08-09:
Hi @doisyg, thank you for this work. I tested this on rectangular footprint robot (polygon footprint) with almost all other param as default. While testing it sometimes randomly terminates with following error:
```
[rviz2-1] [INFO] [1660085482.041054487] [_]: NavigateToPose will be called using the BT Navigator's default behavior tree.           
[bt_navigator-16] [INFO] [1660085482.041549445] [bt_navigator]: Begin navigating from current location to (1.28, -9.03)                                                                                                                           
[controller_server-12] [INFO] [1660085483.292662722] [controller_server]: Received a goal, begin computing control effort.           
[controller_server-12] [ERROR] [1660085483.293591138] [teb_local_planner]: EdgeVelocity::computeError() _error[0]=-nan _error[1]=-nan
[controller_server-12]                                                                                                               
[controller_server-12] terminate called after throwing an instance of 'teb_local_planner::TebAssertionFailureException'              
[controller_server-12]   what():  EdgeVelocity::computeError() _error[0]=-nan _error[1]=-nan                                         
[controller_server-12]                                                                                                               
[ERROR] [controller_server-12]: process has died [pid 18743, exit code -6, cmd '/home/agv_ws/install/nav2_controller/lib/nav2_control
ler/controller_server --ros-args --params-file /tmp/tmpt2exbct1 -r /tf:=tf -r /tf_static:=tf_static -r cmd_vel:=cmd_vel_nav'].       
[lifecycle_manager-19] [INFO] [1660085487.449413561] [lifecycle_manager_navigation]: Have not received a heartbeat from controller_se
rver.   
```
I tried to replicate the conditions but was unsuccessfully. The controller runs perfectly for sometime and then terminates. As his error does not occur on ROS1, I am not sure if this is related to ROS2 port or g2o lib (which is installed by apt).
Do you have any leads what could be causing it?

SteveMacenski, 2022-08-24:
> Who has the maintenance rights nowadays ? I believe a first step should be to create a galactic branch from a copy of the current ros2-master. Then maybe a humble and rolling branch , or we continue to use ros2-master ?

I was given permissions for merging PRs and making new branches, etc but I don't have the time to take on maintaining this codebase. TEB is very complex and I don't have the cycles to add something of this size to my queue. Really, at most I can be a rubber stamp if there's a trusted agent with a well tested PR, but if major changes are made and stuff breaks, I'm not going to really be able to do anything about it. 

I have made a `galactic` branch and the new ros2-master is targeting the API breaking changes for Humble++.

I don't have rights over costmap converter though :frowning:

quantumquasihuman, 2022-09-09:
Hello, has there been any progress regarding this? I would like to use TEB with humble on Ubuntu 22.04,  currently I'm having errors when building from source in galactic branch.

corot, 2022-09-09:
> Hello, has there been any progress regarding this? I would like to use TEB with humble on Ubuntu 22.04, currently I'm having errors when building from source in galactic branch.

I recently run TEB with Humble w/o problems. Please ensure that you are using latest ros2-master branch, as we have recently merged PR #371 that fixes compilation issues

doisyg, 2022-09-09:
Additional, this more complete PR is ready and is going to be merged soon: https://github.com/rst-tu-dortmund/teb_local_planner/pull/374

quantumquasihuman, 2022-09-09:
> I recently run TEB with Humble w/o problems. Please ensure that you are using latest ros2-master branch, as we have recently merged PR #371 that fixes compilation issues

Hi, thanks a lot this helped and fixed all my issues :)

quantumquasihuman, 2022-09-29:
> Hi @doisyg . I checked out your branch (great work btw, thanks) and tested it on a ROS2 docker container -- everything seems to be working as expected.
>
> How can we move forward to make this [...]

Hi, do I understand it correctly that currently Nav2 and TEB don't work together in ROS2 Humble? As stated above I've been having a lot of trouble getting it to work

doisyg, 2022-09-29:
TEB and nav2 are working well together under ros2 Humble (ros2-master branch that you have to compile from source). That was the point of this recent PR: https://github.com/rst-tu-dortmund/teb_local_planner/pull/374

There is not provided launch files for the ROS2 Humble versions and the readme is outdated.

quantumquasihuman, 2022-09-29:
> 
Thank you for this quick reply. Could you point me to a direction (maybe there is a tutorial or more information elsewhere?) so I know how to get it to work for me?

I am new to ros2 and nav2 and don't quite know how to use teb as plugin

tonynajjar, 2022-11-02:
I'm trying to build teb_local_planner from source on Nvidia Xavier Humble (it's Ubuntu 18 but that should not be related to my problem). I cannot install libg2o (ros-humble-libg2o) from apt so I need to build it from source, which I do successfully:

```
root@ebdb5db27d4e:/code/ros2_ws# colcon build --mixin ccache release --packages-select libg2o --symlink-install
Starting >>> libg2o  
Finished <<< libg2o [1.11s]  
```

it's this [tag](https://github.com/ros2-gbp/libg2o-release/tree/release/humble/libg2o/2020.5.29-4/g2o), which I believe is the right one as it is the one installed via apt on Ubuntu 20.

My problem is that when building teb_local_planner, I get:

```
root@ebdb5db27d4e:/code/ros2_ws# colcon build --packages-select teb_local_planner
Starting >>> teb_local_planner
--- stderr: teb_local_planner                             
In file included from /code/ros2_ws/src/teb_local_planner/teb_local_planner/include/teb_local_planner/robot_footprint_model.h:43,
                 from /code/ros2_ws/src/teb_local_planner/teb_local_planner/include/teb_local_planner/teb_config.h:48,
                 from /code/ros2_ws/src/teb_local_planner/teb_local_planner/src/teb_config.cpp:39:
/code/ros2_ws/src/teb_local_planner/teb_local_planner/include/teb_local_planner/pose_se2.h:42:10: fatal error: g2o/stuff/misc.h: No such file or directory
   42 | #include <g2o/stuff/misc.h>
      |          ^~~~~~~~~~~~~~~~~~
compilation terminated.
In file included from /code/ros2_ws/src/teb_local_planner/teb_local_planner/include/teb_local_planner/robot_footprint_model.h:43,
                 from /code/ros2_ws/src/teb_local_planner/teb_local_planner/include/teb_local_planner/teb_config.h:48,
                 from /code/ros2_ws/src/teb_local_planner/teb_local_planner/include/teb_local_planner/visualization.h:45,
                 from /code/ros2_ws/src/teb_local_planner/teb_local_planner/src/visualization.cpp:40:
/code/ros2_ws/src/teb_local_planner/teb_local_planner/include/teb_local_planner/pose_se2.h:42:10: fatal error: g2o/stuff/misc.h: No such file or directory
   42 | #include <g2o/stuff/misc.h>
      |          ^~~~~~~~~~~~~~~~~~
compilation terminated.
In file included from /code/ros2_ws/src/teb_local_planner/teb_local_planner/include/teb_local_planner/robot_footprint_model.h:43,
                 from /code/ros2_ws/src/teb_local_planner/teb_local_planner/include/teb_local_planner/teb_config.h:48,
                 from /code/ros2_ws/src/teb_local_planner/teb_local_planner/include/teb_local_planner/optimal_planner.h:46,
                 from /code/ros2_ws/src/teb_local_planner/teb_local_planner/src/optimal_planner.cpp:42:
/code/ros2_ws/src/teb_local_planner/teb_local_planner/include/teb_local_planner/pose_se2.h:42:10: fatal error: g2o/stuff/misc.h: No such file or directory
   42 | #include <g2o/stuff/misc.h>
      |          ^~~~~~~~~~~~~~~~~~
compilation terminated.
make[2]: *** [CMakeFiles/teb_local_planner.dir/build.make:146: CMakeFiles/teb_local_planner.dir/src/teb_config.cpp.o] Error 1
make[2]: *** Waiting for unfinished jobs....
make[2]: *** [CMakeFiles/teb_local_planner.dir/build.make:118: CMakeFiles/teb_local_planner.dir/src/visualization.cpp.o] Error 1
make[2]: *** [CMakeFiles/teb_local_planner.dir/build.make:90: CMakeFiles/teb_local_planner.dir/src/optimal_planner.cpp.o] Error 1
In file included from /code/ros2_ws/src/teb_local_planner/teb_local_planner/include/teb_local_planner/planner_interface.h:49,
                 from /code/ros2_ws/src/teb_local_planner/teb_local_planner/include/teb_local_planner/homotopy_class_planner.h:58,
                 from /code/ros2_ws/src/teb_local_planner/teb_local_planner/src/homotopy_class_planner.cpp:39:
/code/ros2_ws/src/teb_local_planner/teb_local_planner/include/teb_local_planner/pose_se2.h:42:10: fatal error: g2o/stuff/misc.h: No such file or directory
   42 | #include <g2o/stuff/misc.h>
      |          ^~~~~~~~~~~~~~~~~~
compilation terminated.
make[2]: *** [CMakeFiles/teb_local_planner.dir/build.make:160: CMakeFiles/teb_local_planner.dir/src/homotopy_class_planner.cpp.o] Error 1
In file included from /code/ros2_ws/src/teb_local_planner/teb_local_planner/include/teb_local_planner/robot_footprint_model.h:43,
                 from /code/ros2_ws/src/teb_local_planner/teb_local_planner/include/teb_local_planner/teb_config.h:48,
                 from /code/ros2_ws/src/teb_local_planner/teb_local_planner/include/teb_local_planner/optimal_planner.h:46,
                 from /code/ros2_ws/src/teb_local_planner/teb_local_planner/include/teb_local_planner/teb_local_planner_ros.h:50,
                 from /code/ros2_ws/src/teb_local_planner/teb_local_planner/src/teb_local_planner_ros.cpp:39:
/code/ros2_ws/src/teb_local_planner/teb_local_planner/include/teb_local_planner/pose_se2.h:42:10: fatal error: g2o/stuff/misc.h: No such file or directory
   42 | #include <g2o/stuff/misc.h>
      |          ^~~~~~~~~~~~~~~~~~
compilation terminated.
make[2]: *** [CMakeFiles/teb_local_planner.dir/build.make:174: CMakeFiles/teb_local_planner.dir/src/teb_local_planner_ros.cpp.o] Error 1
/code/ros2_ws/src/teb_local_planner/teb_local_planner/src/recovery_behaviors.cpp:44:10: fatal error: g2o/stuff/misc.h: No such file or directory
   44 | #include <g2o/stuff/misc.h>
      |          ^~~~~~~~~~~~~~~~~~
compilation terminated.
make[2]: *** [CMakeFiles/teb_local_planner.dir/build.make:132: CMakeFiles/teb_local_planner.dir/src/recovery_behaviors.cpp.o] Error 1
[... cut ...]
   42 | #include <g2o/stuff/misc.h>
      |          ^~~~~~~~~~~~~~~~~~
compilation terminated.
make[2]: *** [CMakeFiles/teb_local_planner.dir/build.make:188: CMakeFiles/teb_local_planner.dir/src/graph_search.cpp.o] Error 1
make[1]: *** [CMakeFiles/Makefile2:137: CMakeFiles/teb_local_planner.dir/all] Error 2
make: *** [Makefile:146: all] Error 2
---
Failed   <<< teb_local_planner [6.40s, exited with code 2]
                                
Summary: 0 packages finished [7.09s]
```

Any help would be appreciated.

SteveMacenski, 2022-11-02:
https://github.com/ros/rosdistro/blob/master/humble/distribution.yaml#L2097 this is the version in rosdep, you might want to see if those are the same.

tonynajjar, 2022-11-03:
it is indeed: 

```
root@ebdb5db27d4e:/opt/ros/humble/src/libg2o# git status
HEAD detached at release/humble/libg2o/2020.5.29-4
```

I also edited the building logs in my first post to show that libg2o was in fact found.

SteveMacenski, 2022-11-03:
If you look at your install, can you find that header?

tonynajjar, 2022-11-03:
```
root@ebdb5db27d4e:/code/ros2_ws# ls install/libg2o/include/g2o/stuff/
color_macros.h  filesys_tools.h  macros.h  opengl_primitives.h  os_specific.h  sampler.h        string_tools.h  timeutil.h     unscented.h
command_args.h  g2o_stuff_api.h  misc.h    opengl_wrapper.h     property.h     sparse_helper.h  tictoc.h        tuple_tools.h
```

you're referring to the header from the error right? -> misc.h. If so then yes it's there

SteveMacenski, 2022-11-03:
Is that simply the first header in the files that its complaining about from g2o? Maybe it has nothing to do with that file, but just finding _any_ file in g2o.

tonynajjar, 2022-11-03:
> Maybe it has nothing to do with that file, but just finding any file in g2o.

Yes I think this was the problem. I built and installed g2o with `make install` instead of colcon and now it works. I still don't know what's the difference though, I didn't pass any extra arguments to the `make` workflow.

corot, 2024-11-10:
Afaik all works now on both humble and jazzy, so we can close this

closed 2024-11-10 by corot