rst-tu-dortmund/teb_local_planner #380: When I give way with via_point it constantly follows the path

mduz22, 2022-08-16:
Hi! 

I have a Differential Drive robot and I am using TEB Local planner for local and global planner. I want to draw my own path by giving via points and I am sending certain points to the topic /move_base/TebLocalPlannerROS/via_points. It follows the points, but when it reaches the last point, it wants to return to the first point again and creates an endless loop. I don't want the robot to do that. I want it to stop when it gets to the last via_point. How can I solve this problem. 

Is there a way to do this?

System:
Ubuntu 20
ROS Noetic

[image]

eithwa, 2022-09-20:
I have the same problem, and I found that we need to edit the code
https://github.com/rst-tu-dortmund/teb_local_planner/blob/0e839074c3407ff9ee2206a49567a20bd49fd8cc/src/teb_local_planner_ros.cpp#L295
```
  int via_size = custom_via_points_active_?0:via_points_.size();
  if(fabs(std::sqrt(dx*dx+dy*dy)) < cfg_.goal_tolerance.xy_goal_tolerance
    && fabs(delta_orient) < cfg_.goal_tolerance.yaw_goal_tolerance
    && (!cfg_.goal_tolerance.complete_global_plan || via_size == 0)
    && (base_local_planner::stopped(base_odom, cfg_.goal_tolerance.theta_stopped_vel, cfg_.goal_tolerance.trans_stopped_vel)
        || cfg_.goal_tolerance.free_goal_vel))
  {
    goal_reached_ = true;
    return mbf_msgs::ExePathResult::SUCCESS;
  }
```

mduz22, 2022-09-20:
> I have the same problem, and I found that we need to edit the code, not the best way, but you can make do with it
> 
> https://github.com/rst-tu-dortmund/teb_local_planner/blob/0e839074c3407ff9ee2206a49567a20bd49fd8cc/src/teb_local_planner_ros.cpp#L295
> 
> ```
>   int via_size = custom_via_points_active_?0:via_points_.size();
>   if(fabs(std::sqrt(dx*dx+dy*dy)) < cfg_.goal_tolerance.xy_goal_tolerance
>     && fabs(delta_orient) < cfg_.goal_tolerance.yaw_goal_tolerance
>     && (!cfg_.goal_tolerance.complete_global_plan || via_size == 0)
>     && (base_local_planner::stopped(base_odom, cfg_.goal_tolerance.theta_stopped_vel, cfg_.goal_tolerance.trans_stopped_vel)
>         || cfg_.goal_tolerance.free_goal_vel))
>   {
>     goal_reached_ = true;
>     return mbf_msgs::ExePathResult::SUCCESS;
>   }
> ```

Thank you for your info. Did you try to edit this code? Because it seems a bit complex.

eithwa, 2022-09-20:
I've tried it, but still some problems, I'm still looking for the real reason

mduz22, 2022-09-20:
Actually, I've solved it in a simple way for now but it is not a real solution. I send an empty array to the via_points topic when it approaches the point I want according to the data I get from my odom topic. Looks like it solved my problem for now.

eithwa, 2022-09-20:
Clearing via_point does work, but I can't do this on my usage
, I will still find a solution, thank you

mduz22, 2022-09-22:
Hey, How is it going for you? Could you set this code?

eithwa, 2022-09-28:
For my use, I don't need to make the robot fit the via_point completely, so I modified the code to only judge the via_point within 1 meter of the robot

closed 2022-12-05 by mduz22