rst-tu-dortmund/teb_local_planner #384: teb_local_planner process died

apeng-yp, 2022-09-26:
when i use teb_local_planner,  this has some error sometimes. this prints is   [teb_local_planner]: EdgeVelocity::computeError() _error[0]=nan _error[1]=nan

automech-rb, 2022-10-06:
Hi @apeng-yp 
I had similar error and found that it usually occur with planner creating in-place rotation.
This is because deltaT estimate is zero on following line.
https://github.com/rst-tu-dortmund/teb_local_planner/blob/0e839074c3407ff9ee2206a49567a20bd49fd8cc/include/teb_local_planner/g2o_types/edge_velocity.h#L106

Can you let me know which planner you are using?

apeng-yp, 2022-10-12:
Hi @automech-rb  
Thank you for your attention

apeng-yp, 2022-10-12:
Hi @automech-rb
Thank you for your attention, i use the nav2_navfn_planner/NavfnPlanner and nav2_straightline_planner/StraightLine,  all have the same problem。

apeng-yp, 2022-10-12:
Hi @automech-rb
What kind of planner were you using？ Have you solved or circumvented the problem？