rst-tu-dortmund/teb_local_planner #399: The result of TEB calculation is very abnormal?

xianglunkai, 2023-01-31:
[image]

xianglunkai, 2023-01-31:
some warning:
[ WARN] [1675151857.924004555]: Control loop missed its desired rate of 10.0000Hz... the loop actually took 0.2111 seconds
[ WARN] [1675151857.928814744]: Map update loop missed its desired rate of 10.0000Hz... the loop actually took 0.1791 seconds

xianglunkai, 2023-01-31:
TebLocalPlannerROS:

 odom_topic: /odom
 map_frame: odom_combined

 # Trajectory
 teb_autosize: True #优化期间允许改变轨迹的时域长度
 dt_ref: 0.3 #局部路径规划的解析度# minimum 0.01
 dt_hysteresis: 0.1 #允许改变的时域解析度的浮动范围， 一般为 dt_ref 的 10% 左右 minimum0.002
 global_plan_overwrite_orientation: False #覆盖全局路径中局部路径点的朝向
 max_global_plan_lookahead_dist: 3.0 #考虑优化的全局计划子集的最大长度
 feasibility_check_no_poses: 3 #检测位姿可到达的时间间隔 minimum 0

 # Robot
 max_vel_x: 0.4 #最大x前向速度
 max_vel_y: 0 #最大y前向速度，非全向移动小车需要设置为0
 max_vel_x_backwards: 0.4 #最大后退速度
 max_vel_theta: 0.3  #最大转向角速度
 acc_lim_x: 0.2  #最大x向加速度
 acc_lim_y: 0  #最大y向加速度，非全向移动小车需要设置为0
 acc_lim_theta: 0.3  #最大角加速度

 #阿克曼小车参数，非阿克曼小车设置为0
 min_turning_radius: 0.1   #for not_akm    # Min turning radius of the carlike robot (compute value using a model or adjust with rqt_reconfigure manually)
 wheelbase: 0.5   #for not_akm
 
 cmd_angle_instead_rotvel: False #无论是不是阿克曼小车都设置为false，因为阿克曼小车启用了阿克曼速度转换包
                                 #true则cmd_vel/angular/z内的数据是舵机角度

 footprint_model: # types: "point", "circular", "two_circles", "line", "polygon" 默认“point”类型
  type: "polygon"   #多边形类型for akm/mec，默认”point”。
  #type: "circular" #多边形类型for omni
  #line_start: [0.00, 0.0] # for type "line" senior_akm
  #line_end: [0.7, 0.0] # for type "line" top_akm_bs
  #radius: 0.3 # for type "circular"
  #front_offset: 0.2       #for type "two_circles"
  #front_rasius: 0.2       #for type "two_circles"
  #rear_offset : 0.2       #for type "two_circles"
  #rear_rasius : 0.2       #for type "two_circles"

  #差速系列
  # vertices: [[-0.1350, -0.1110], [-0.1350, 0.1110], [0.1350, 0.1110], [0.1350, -0.1110]] #多边形端点坐标 for mini_4wd
  vertices: [[-0.62, -0.38], [-0.62, 0.48], [0.40, 0.48], [0.40, -0.38]] #多边形端点坐标 for mini_4wd
 xy_goal_tolerance:  0.4   #目标 xy 偏移容忍度  minimum 0.001 maximum 0.2
 yaw_goal_tolerance: 0.2 #目标 角度 偏移容忍度 minimum 0.001 maximum 0.1
 free_goal_vel: False  #允许机器人以最大速度驶向目的地
 complete_global_plan: True
 
 # Obstacles
 min_obstacle_dist: 0.1 #和障碍物最小距离
#  inflation_dist: 0.05
 include_costmap_obstacles: True  #是否将动态障碍物预测为速度模型，
 costmap_obstacles_behind_robot_dist: 1.5  #限制机器人后方规划时考虑的局部成本地图障碍物
 obstacle_poses_affected: 15   #障碍物姿态受影响0~30
 costmap_converter_plugin: ""
 costmap_converter_spin_thread: True
 costmap_converter_rate: 5
 include_dynamic_obstacles: True 
 dynamic_obstacle_inflation_dist: 0.6 # 0.6

 # Optimization  
 no_inner_iterations: 5
 no_outer_iterations: 4
 optimization_activate: True
 optimization_verbose: False
 penalty_epsilon: 0.1
 obstacle_cost_exponent: 4
 weight_max_vel_x: 2
 weight_max_vel_theta: 1
 weight_acc_lim_x: 1
 weight_acc_lim_theta: 1
 weight_kinematics_nh: 1000
 weight_kinematics_forward_drive: 1
 weight_kinematics_turning_radius: 1
 weight_optimaltime: 1 # must be > 0
 weight_shortest_path: 0
 weight_obstacle: 100
 weight_inflation: 0.2
 weight_dynamic_obstacle: 10 # not in use yet
 weight_dynamic_obstacle_inflation: 0.2
 weight_viapoint: 1
 weight_adapt_factor: 2

 # Homotopy Class Planner
 enable_homotopy_class_planning: False
 enable_multithreading: True
 max_number_classes: 4
 selection_cost_hysteresis: 1.0
 selection_prefer_initial_plan: 0.95
 selection_obst_cost_scale: 1.0
 selection_alternative_time_cost: False

 roadmap_graph_no_samples: 15
 roadmap_graph_area_width: 5
 roadmap_graph_area_length_scale: 1.0
 h_signature_prescaler: 0.5
 h_signature_threshold: 0.1
 obstacle_heading_threshold: 0.45
 switching_blocking_period: 0.0
 viapoints_all_candidates: True
 delete_detours_backwards: True
 max_ratio_detours_duration_best_duration: 3.0
 visualize_hc_graph: False
 visualize_with_time_as_z_axis_scale: False

 # Recovery
 shrink_horizon_backup: True
 shrink_horizon_min_duration: 10
 oscillation_recovery: False
 oscillation_v_eps: 0.1
 oscillation_omega_eps: 0.1
 oscillation_recovery_min_duration: 10
 oscillation_filter_duration: 10

skstudying, 2023-03-02:
我遇到了一样的问题，请问你找到问题出现的原因了吗？

closed 2023-06-02 by xianglunkai