rst-tu-dortmund/teb_local_planner #403: TEB for articulated center steering

zonared, 2023-02-14:
This is not so much an issue but is really a question of how to modify this planner so it support center articulation (like a front-end loader )

How can the planner be modified to support such a robot?

[image]
[image]

Thanks in advance, David...

lijiaming5550, 2025-02-07:
Hello, how is the model of your articulation structure created, and is there any relevant information you can recommend? At present, I am also working on the planning algorithm of articulated structures, and we can communicate more。

zonared, 2025-02-10:
Hi there, thanks for commenting on my '_issue_'. :)

Ok, firstly I built the physical machine out of parts. Then to make rviz look better and help me visualize the robot easier, I painstakingly measured all the part and manually modelled the robot using Solidworks, then I used URDF exporter to export out of Solidworks.

As for some info I can recommend, that depends on what you've tried and your robot. For mine, I spent ages randomly tuning different parameters of TEB, nothing really worked well. With my robot the 'center' which is base_link, is the articulation steering pivot. The problem with this is, every time you change the angle, base_link shifts sideways in the Y axis. What this does is shift the robot position away from the TEB planned path and as soon as that happens TEB will plan a new path. So, this creates a perpetual loop of replanning which results in oscillations of the steering angle and never following path.

My robot takes ~10seconds to articulate from lock to lock, steering angle roughly -0.45rad to +0.45rad, so it steers slowly and not every far. So, if you're moving and steering at the same time, it makes control tricky, but it's necessary for efficiency.

I also experimented with translating my base_line back from the center to the middle of the rear wheels, this helped a little with TEB because steering didn't affect the planned path too much due to less sideway drift, but still didn't produce smooth and efficient path following. The best I could get was navigation with MANY steering oscillations and forward backward movements (inefficient), but it would eventually reach the goal given enough time (I have a YouTube video if you interested, PM for link).

I am now experimenting DRL by [Reinis Cimurs](https://medium.com/@reinis_86651/deep-reinforcement-learning-in-mobile-robot-navigation-tutorial-part1-installation-d62715722303), but this is VERY difficult for me as I don't have any formal training on deep learning.

What can you tell me about your robot and any progress you've had?