rst-tu-dortmund/teb_local_planner #410: undefined reference to `int boost::math::sign<double>(double const&)'

muellerbernd, 2023-06-05:
Hello, on archlinux and NixOS we get an error while compiling: `undefined reference to int boost::math::sign<double>(double const&) `. The reason is that in newer boost versions you need to include a new header. We have fixed that with the following patch:
```patch
diff --color -Naur teb_local_planner-release-release-noetic-teb_local_planner/include/teb_local_planner/h_signature.h teb_local_planner-release-release-noetic-teb_local_planner_new/include/teb_local_planner/h_signature.h
--- A/include/teb_local_planner/h_signature.h   2020-05-29 18:12:46.000000000 +0200
+++ B/include/teb_local_planner/h_signature.h   2023-01-20 19:56:56.259209108 +0100
@@ -51,6 +51,7 @@
 #include <functional>
 #include <vector>
 #include <iterator>
+#include <boost/math/special_functions/sign.hpp>

 namespace teb_local_planner

```
see also [this issue](https://github.com/lopsided98/nix-ros-overlay/issues/266) in the nix-ros-overlay repo.

loco-choco, 2023-06-28:
Created this fix so I could use it in my projects.

renan028, 2026-01-09:
https://github.com/rst-tu-dortmund/teb_local_planner/pull/431

closed 2026-01-09 by renan028