rst-tu-dortmund/teb_local_planner #432: unexpected steer angle with car like robot

imranhussain14, 2025-05-19:
Hello ia m using the teb local planner for navigation in orchard with a car like robot. My robot have a mix ackermann steering so i am using belo configuration:

TebLocalPlannerROS:
  odom_topic: /rtabmap/odom
  map_frame: odom
  # Robot configuration (kinematics)
  acc_lim_x: 0.5       
  acc_lim_theta: 0.5   
  max_vel_x: 0.1      
  max_vel_x_backwards: 0.05  
  max_vel_theta: 0.12  
  min_vel_x: 0.05       
  min_vel_theta: 0.05  

  # Ackermann steering constraints
  cmd_angle_instead_rotvel: false 
  wheelbase: 1.14               
  min_turning_radius: 1.7        #

  # Footprint model
  footprint_model:
    type: "point"              
    front_offset: 0.57               
    front_radius: 0.6915            
    rear_offset: -0.57              
    rear_radius: 0.6915              
    radius: 0.8636                 
    line_start: [-1.2948, 0.0]       
    line_end: [1.2948, 0.0]          
    vertices: [[1.1985, -0.06915], [0.86292, -0.06915], [0.86292, -0.24894], [-0.90882, -0.24894], [-0.90882, 0.0], [-0.90882, 0.24894], [0.86292, 0.24894], [0.86292, 0.06915], [1.1985, 0.06915]] # UNCHANGED: Kept for polygon model
  # Goal tolerance
  xy_goal_tolerance: 0.2    
  yaw_goal_tolerance: 0.1  
  free_goal_vel: false     

  # Trajectory configuration
  dt_ref: 0.3              
  dt_hysteresis: 0.1    
  max_samples: 500       
  global_plan_overwrite_orientation: true 
  allow_init_with_backwards_motion: false  
  max_global_plan_lookahead_dist: 3.0  
  force_reinit_new_goal_dist: 1.0     
  feasibility_check_no_poses: 5       

  # Obstacle avoidance
  min_obstacle_dist: 0.8    
  inflation_dist: 0.6     
  dynamic_obstacle_inflation_dist: 0.6  
  include_dynamic_obstacles: true     
  obstacle_poses_affected: 30          
  costmap_obstacles_behind_robot_dist: 1.0  

  # Optimization weights
  weight_max_vel_x: 2.0     
  weight_max_vel_theta: 1.0   
  weight_acc_lim_x: 1.0     
  weight_acc_lim_theta: 1.0  
  weight_kinematics_turning_radius: 1.0 
  weight_kinematics_forward_drive: 1000.0  
  weight_optimal_time: 1.0    
  weight_obstacle: 50.0      
  weight_inflation: 10.0      
  weight_dynamic_obstacle: 10.0  
  weight_viapoint: 1.0       
  penalty_epsilon: 0.01

  # Homotopy class planning
  enable_homotopy_class_planning: true 
  enable_multithreading: true         
  max_number_classes: 4                
  selection_cost_hysteresis: 1.0    
  selection_obst_cost_scale: 100.0      
  selection_alternative_time_cost: false 

  # Recovery behaviors
  shrink_horizon_backup: true  
  oscillation_recovery: true    
  oscillation_filter_duration: 10.0  

The problem is that even teh apth is stright but when i look for the ackerman_cmd it gives me wrong a steer angle continuously which makes a bad turn in my navigation.:

[image]

[image]
With this unexpected steer behaviur my robot cannot follow global plan perfectly, anyone having experience with car like robot can help me?

OYSJ1997, 2025-08-12:
> [@croesmann](https://github.com/croesmann) can you give me an update regrading this?
I also encountered the problem of the Ackerman car swaying from side to side when driving. Can you tell me which parameter settings will have this effect?

imranhussain14, 2025-08-12:
@OYSJ1997 I am not sure whats your configuration but in my case, I was using "cmd_angle_instead_rotvel: false" and then i am not converting it into properly angle in my base node. the cmd_vel (angular.z) is in rad/sec in this case. and you need to convert the Twist Message into Ackermann Message if you are using Ackermanns.